don/BluetoothSerial #186: Connection to Adafruit Flora Bluefruit LE failing

zpeterg, 2016-01-10:
I am unable to connect to a Flora 2 via Bluefruit LE (flora edition, which is a version of Adafruit Friend). I am using Adafruit's BleUart example (https://learn.adafruit.com/adafruit-flora-bluefruit-le/bleuart) and successfuly communicating with their Bluefruit LE Android app, but I cannot connect with either bluetoothSerial.connect or bluetoothSerial.connectInsecure. I am using the MAC address returned by bluetoothSerial.list. Here is the result of .list:

address: "EC:FA:F4:44:31:79"
class: 7936
id: "EC:FA:F4:44:31:79"
name: "Adafruit Bluefruit LE"
__proto__: Object
length: 2
__proto__: Array[0]

Here is an example of the connection attempt that keeps failing:

bluetoothSerial.connectInsecure('EC:FA:F4:44:31:79', function(){console.log('connected'); }, function(){console.log('not connected'); });

I am using a MotoX 1st Gen running Android 4.4.4.

Any help would be much appreciated as I have gone round-and-round on this with no success!

don, 2016-01-10:
@zpeterg this plugin only supports Bluetooth Classic for Android, so you won't be able to connect to the Bluefruit LE.

You need [cordova-plugin-ble-central](https://github.com/don/cordova-plugin-ble-central). There's an example project that connects to the Adafruit devices running the Nordic UART service. See https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle

zpeterg, 2016-01-12:
It works! Thank you very much!

closed 2016-01-15 by don