don/BluetoothSerial #187: Bluetooth Connection via WP8 fails

Droggel, 2016-01-11:
I was trying to run the exsamples from the Bluetooth serial plugin, and always getting this error message while compiling: 

cordovalib\NativeExecution.cs(127,17): warning CS4014: Da auf diesen Aufruf nicht gewartet wird, wird die Ausführung der aktuellen Methode vor Abschluss des Aufrufs fortgesetzt. Ziehen Sie ein Anwenden des 'Await'-Operators auf das Ergebnis des Aufrufs in Betracht. [C:\Users\meed_rots\Desktop\counter\platforms\wp8\Counter.csproj]

Means that at line 127 the operation will be continued without waiting to the end of the actual method. The "Await" Operation maybe has to be used (See Microsoft link, too).

https://msdn.microsoft.com/de-de/library/hh873131.aspx

This error happens on every of the 4 exsamples always exactly the same way and all aplications are crashing directly after start and first change of a value which sould be send. 

Complete "Test" i did (for LED example):

- create a new cordova project: cordova create led com.example.led Led
- copy all example code in the new project forlder
- add plugin: cordova plugin add cordova-plugin-bluetooth-serial
- add platform: cordova platform add wp8
- prepare: cordova prepare wp8
- compile as debug: cordova compile wp8 --debug

error massage is shown but build is successful.

open the .sln file in Visual Studio Same warning is shown in the error list.

- Run in debug mode on device
- Bluetooth connection to every device is getting always lost by starting the app (tried with using a HC-06 Bluetooth device for arduino, tested with different software for WP8, works fine; tried with pairing to a win 10 PC using putty, too)
- LED app is crashing if slider value is getting changed.

I dont know how to use this await command correctly so I would be really thankful for any suggestion. 

Equipment:
cordova: Ver: 5.4.1
Visual Studio Ultimate 2012, Version 11.0.61030.00 Update 4
Windows Phone 8.0 SDK
Phone: Nokia Lumia 920, Windows Phone 8.1 Update Version 8.10.14234.375

Trying the same example for Android is working fine but unfortunately wp8 is important for me ;) 

Best Regards 
Steffen (sorry for maybe bad english but i try my best ;) )

Droggel, 2016-01-11:
[wp8.zip](https://github.com/don/BluetoothSerial/files/85845/wp8.zip)

don, 2016-01-15:
@Droggel my Visual Studio setup is broken. It will be a bit before I can duplicate this.

Droggel, 2016-01-18:
ok but thank you very much anyway that you are taking care about it :)

Droggel, 2016-02-24:
Hi, I am sorry for request but do you found any solution for the Problem because I did not so far :S

don, 2017-03-19:
@Droggel sorry I never got a chance to look into this. Since it was over a year ago, I'm assuming you found a solution or abandoned this. If not, comment here and I'll re-open the issue.

closed 2017-03-19 by don