don/BluetoothSerial #189: Read / write data through connection with a HID device

xjxxjx1017, 2016-01-13:
Hi, I'm Jackson. we are working on a project using a HID device ( writing pad ) as an input. The original SDK for the writing pad does not support Cordova/Phonegap, but is available for Android, IOS and Chrome app/plugin ( https://github.com/kent-displays ). So we try to use BluetoothSerial plugin to communicate with the device.

We successfully connected to the HID device by using bluetoothSerial.connectInsecure(), but we were not able to receive any data from the device input. I wonder whether it's because that bluetooth serial and bluetooth HID are different things and shall be handled differently.

don, 2016-01-15:
@xjxxjx1017 if you're just looking to do [HID](https://developer.bluetooth.org/TechnologyOverview/Pages/HID.aspx) or [HID over GATT](https://developer.bluetooth.org/TechnologyOverview/Pages/HOGP.aspx) this plugin won't help.

This plugin supports [SPP](https://developer.bluetooth.org/TechnologyOverview/Pages/SPP.aspx) for Android and UART BLE services (that behave like SPP) for iOS.

It's possible that the hardware you have simultaneously supports HID and other Bluetooth protocols. I'm not sure what you're trying to do with the hardware.

If it was my projects, I'd look into writing a new Cordova plugin that supports the hardware. Since the vendor provides iOS and Android implementations they've done a lot of the hard work for you.

xjxxjx1017, 2016-01-16:
Thanks you very much Don, I've been writing a new plugin this week, which works well. Thanks for the suggestion.

closed 2016-01-16 by xjxxjx1017