don/BluetoothSerial #192: Android to android bluetooth communication

RobinDesir, 2016-01-14:
Hey Don, 

Is it possible to communicate with another Android device ? I keep getting the "Unable to connect to device" error ... 

Thx in advance for your answer, 
Robin.

RobinDesir, 2016-01-15:
I read that it was because Android to Android communication via this plugin wasn't supported in another issue, is it still true ?

don, 2016-01-15:
@RobinDesir Android to Android isn't supported by default. If you see Issue #50 some people have android-to-android working in forks of the plugin.

Ideally once someone has this working and tested, they'll submit a pull request so it can be part of the library.

Closing this as a duplicate of #50

closed 2016-01-15 by don

RobinDesir, 2016-01-15:
Thx for your quick answer don.

commit 8418489a (2015-09-18): Drop support for Cordova 2.9 fixes #192