don/BluetoothSerial #193: Print list of bluetooth devices to dom

mumairali, 2016-01-22:
I am new and after spending half day but didn't get solution. I am trying to list devices into DOM. later on when user click on device then phone connected to that device. I use the following code to list.

    `$("#listBTDevices").on('click', function() {
            bluetoothSerial.list(
                function(devices) {
                        devices.forEach(function(device) {
                            $("#result").append('<p>'+ device.name +'  ---  '+ device.id +'</p>');
                });
            }, 
            function() {
                    $("#result").text("No device available");
            }
        );
    }); `

This code run successfully but it disappears all the devices after its loop complete. Also I tried to return device but it gives undefined. I not sure what I am doing wrong. Can you please show me exact way to achieve this? my whole work is stuck

don, 2016-01-22:
See [onDeviceList](https://github.com/don/BluetoothSerial/blob/master/examples/Chat/www/js/index.js#L44) in the [Chat example](https://github.com/don/BluetoothSerial/tree/master/examples/Chat) for code that adds items to a list.

mumairali, 2016-01-22:
Thanks given example help me alot !

closed 2016-01-26 by don