don/BluetoothSerial #194: Add support for Android 6.x

driannaude, 2016-01-27:
Great work so far, this may transform into a PR at some stage, but thought I'd file an issue in the interrim.  There seems to be issues with Permissions on Android 6.x (Marshmellow) looking at the logcat dump:

```shell
01-28 07:19:59.027  2182  2182 V BluetoothDiscoveryReceiver: Received: android.bluetooth.adapter.action.DISCOVERY_STARTED
01-28 07:20:03.018  2209  2235 D bt_btif_config: btif_get_device_type: Device [d5:c9:3a:1b:6b:32] type 2
01-28 07:20:03.025   608   621 W BroadcastQueue: Permission Denial: receiving Intent { act=android.bluetooth.device.action.FOUND flg=0x10 (has extras) } to ProcessRecord{7f5bc1c 1751:net.enatel.vhfmconnect/u0a166} (pid=1751, uid=10166) requires android.permission.ACCESS_COARSE_LOCATION due to sender com.android.bluetooth (uid 1002)
01-28 07:20:08.054  2209  2333 E bt_btif : bte_scan_filt_param_cfg_evt, 1
```

FYI I've cleaned up the dump and remvoed all the noise to isolate the BT specific messages.

Running on a **Nexus 7**
Android **6.0.1**
npm  **3.3.2**
node **5.5.0**
cordova **5.4.1

don, 2016-01-27:
I need to add Android 6 support.  In the mean time target android-22 to avoid this error.

don, 2016-04-20:
@driannaude which function(s) were you calling that causes the permission error?

driannaude, 2016-04-25:
@don, I should've probably mentioned this a while ago, turns out that coarse location permissions are required for bluetooth on 6.x. It's not very well known/documented in the Android spec so I wrongly assumed it must be the plugin. Sorry about that! So requesting the ACCESS_COARSE_LOCATION permission seemed to have fixed the issue. Perhaps this isn't a total loss though. The plugin could add a console warning if the android version is 6.x and the location permissions have not been requested?

Hope this helps someone in the future though!

I'd say we can close this off.

commit 408a03a0 (2016-09-17): Add support for Android 6.0 (API 23) permissions

Fixes #194, Fixes #226
This is a different implementation of @Sundypha's pull request #228
closed 2016-09-17 by don