don/BluetoothSerial #196: WP8.1 Missing Command Error

mareksip, 2016-01-29:
Hello, after executing 

    bluetoothSerial.isEnabled(onSuccess, onFailure);

I receive message: **Missing Command Error**. I am debugging directly from VS2015 on my HTC 8S device.

Does anyony has solution to this issue?

Thanks,
Marek

don, 2016-01-29:
Make sure you're running your app as WP8 (Silverlight) and not the newer 8.1 or 10.

mareksip, 2016-01-29:
@don Thanks for response, is there any plan to support 8.1 or 10? I might help with it, if it can be done..

don, 2016-01-29:
Yes, I'd like to support both 8.1 and 10. It can be done, but I haven't looked into the details of what's required to port. If you'd like to help, create a fork of the and give it a try.

closed 2017-03-17 by don