don/BluetoothSerial #197: Plugin referencing issue on Android API 23

mareksip, 2016-02-04:
I have tried to add the plugin on Android API 23 project and after compilation and deployment on device I got  

`bluetoothSerial undefined error`

I have looked into the references after adding the project and found out that all of the other plugins included in `platforms/android/android.json` are added to `parents` node, only bluetoothSerial was added to `/widgets` so I changed the plugin.xml from
`<config-file target="res/xml/config.xml" parent="/widget">`

to

`<config-file target="res/xml/config.xml" parent="parents">`

And it is working, this might need further investigation, it is not clear to me what is happening.

don, 2016-02-07:
@mareksip thanks for submitting this. I guess the config.xml changed with the latest android. I'll look into this and get updated rolled into the next release soon.

don, 2016-04-20:
@mareksip I can't duplicate this using Cordova 6.1.1 and Android 6.0.1

I'm guessing some updates to Cordova fixed everything

closed 2016-04-20 by don