don/BluetoothSerial #199: subscribeRawData does not return anything

mikecunneen, 2016-02-16:
I'm using v. 0.4.5 of the plugin on an Ionic 1 App. My device sends back binary data so I am trying to use subscribeRawData() but it refuses to return any data. 

All the other functions work fine and my device tells me I am subscribed successfully. The regular subscribe() functions returns data but it's a string and I can't seem to parse it into anything else.

The attached screenshot is the data from the read function and as you can see it's unreadable.

I'm really stuck with this and would really appreciate some advice.

Thanks for an awesome plugin :)

Mike
[image]

don, 2016-02-17:
@mikecunneen I modified the [Counter example](https://github.com/don/BluetoothSerial/tree/master/examples/Counter) to print the subscribed data and the raw data. Everything seemed to work OK.

Can you try on your machine?

Add a div to the HTML for the raw data to index.html

        <div id="rawDataDiv"></div>

Replace index.js from the example with https://gist.github.com/don/a8fb0b9b11a4393e4b9b. Be sure to update the mac address var to match your hardware.

The change subscribe for raw data in onConnect

    bluetoothSerial.subscribeRawData(app.onRawData, app.subscribeFailed);

And add a function to handle the raw data callbacks

    onRawData: function(data) {
        rawDataDiv.innerHTML = new Uint8Array(data);
    },

I was using an Arduino with the [Counter sketch](https://github.com/don/BluetoothSerial/blob/master/examples/Counter/Arduino/Counter/Counter.ino) but you should be able to use your existing device, since the UI just prints out the data it receives.

The data comes back as an ArrayBuffer, so you need to create a typed array before you can use it. If it's bytes, I usually use a Uint8Array. If you're expecting different data use what makes sense for your data e.g. Float32Array.

mikecunneen, 2016-02-18:
Hi Don,

That's very much for the detailed advice. I will try this now and let you know how it goes.

Thanks,

Mike

itchix, 2016-02-29:
Hi @mikecunneen ,

Is it possible to get your code for the subscribe of your ionic app, because I don't understand why my subscribe function doesn't call the callback...

Thanks

mikecunneen, 2016-02-29:
Hi Don,

I totally forgot to reply to you last week, sorry. 

Your counter example worked for me :)

 ```
 bluetoothSerial.subscribeRawData(onRawData, subscribeFailed);

                                            // this runs for every byteArray.... could run 5-25 times...
                                            function onRawData(data)
                                            {
                                              
                       var rawData = new Uint8Array(data);
                                                  //alert("rawData (json) is: " + JSON.stringify(rawData) );
                                                    var dataArrayJSON = JSON.stringify(rawData);

                                                    for (var e = 0; e < rawData.length; e++)
                                                    {
                                                  //thisBPData is an array that combines all the byteArrays from my device.
                                                    thisBPData.push(rawData[e]);
                                                    }
                                             }
```
Thanks again for your help Don, and the awesome plugin!!

Mike

closed 2016-02-29 by mikecunneen