don/BluetoothSerial #202: try to get a return value of bluetoothSerial.readUntil

Sega01, 2016-02-22:
Hi, 

i try to get a return value of that following function, but everything i get is undefined, someone have an idea?

`bluetoothSerial.readUntil('\n', function (data) {
    console.log(data);
}, failure); `

don, 2016-02-22:
Your callback is begin called with data being undefined?

Are you sending binary data delimited by \n?

Maybe [https://github.com/don/BluetoothSerial#subscriberawdata](bluetoothSerial.subscribeRawData) is a better fit? With subscribeRaw you get data in the callback whenever messages arrive on iOS, or basically continuously on Android.

Sega01, 2016-02-22:
Ah misswriting by me there, sorry:

Im getting datas through the Bluetooth, 1 or 0, like i want.
I only want to change another variable in my Ionic app depending what i get through the bluetooth, or better in the phaser.js game that is inside the app.
When im trying to combine bluetooth and phaser the game crashes, so i thought i could return the value of that function and save it to an external value?

closed 2016-02-23 by Sega01