don/BluetoothSerial #206: ipod: Some devices not shown with list() but are seen in settings / bluetooth menu?

sclearion, 2016-03-14:
The plugin works fine but having some issues: 

- my Zebra printers (BT 3.0) show in **android version of app** using ```setDeviceDiscoveredListener()``` just fine.
- when using ```list()``` in my **ios version of app** I do not see any of my printers, only BLE devices that I recognize. but no zebra printers

- I then close my **ios version of app**, go to the Settings menu on the same ios device and see that the phone **can list** and **pair to** my zebra printers, but ```list()``` cannot. How do I bridge this gap or can I add the UUID of the printers manually to ```BLEDefines.h```

closed 2016-03-14 by sclearion

JMGunnels3, 2016-03-24:
Hey @sclearion, I'm having a similar issue with ios not listing devices that show up on android. Did you find a solution to this?

sclearion, 2016-03-24:
Nope, we gave up. We just ordered new BTLE printers rather than fork the
library.

It's important to note that this library IS NOT designed to mimic the
native Bluetooth display shown in iOS -> Settings -> Bluetooth!!!!

From the looks of it and the issue stream, this project was design so
someone could connect via Bluetooth with specific project (arduino, etc)
devices.

As it stands, there IS NO ionic/cordova library that exhibits the native
ios bluetooth api (which will show BT3.0 and BT4.0+).

don, 2016-03-24:
@JMGunnels3 On Android this library works with Bluetooth Classic devices that support SPP.

iOS works differently. On iOS this library uses Bluetooth Low Energy. It supports devices that are running  one of 5 select SPP-like UART services. The BLE support in this library is very focused. If you need generic BLE support on iOS and Android try https://github.com/don/cordova-plugin-ble-central.

 iOS doesn't support Bluetooth Classic unless you're part of the Apple MFi program.

sclearion, 2016-03-24:
Ios does support bluetooth classic. This has been verified. My company Suntech has 125 devices in field and dedicated apple support and they have verified that ios WILL connect to BT devices and we are currently. This is why you see older BT devices in the ios settings but not with the library

sclearion, 2016-03-24:
The library work fine with android but you need to use 
https://github.com/don/cordova-plugin-ble-central

don, 2016-03-25:
@sclearion You're right. I should have stated this more clearly.

iOS supports a limited number of Bluetooth Classic profiles. https://support.apple.com/en-us/HT204387

Classic Bluetooth **data** connections on iOS require a MFi license and royalties to Apple. https://developer.apple.com/programs/mfi/

So, you can pair a Bluetooth Classic device with iOS, but you can't use [Serial Port Profile (SPP)](https://developer.bluetooth.org/TechnologyOverview/Pages/SPP.aspx).