don/BluetoothSerial #207: Image printing support

ivangutierrezr, 2016-03-14:
Hi, Do you have some image print support in your plugin?. I have to print a logo and barcode with a bixolon printer. For text it works fine to me but i need print images, can you help me?

sclearion, 2016-03-14:
If it is ZPL, check out the docs http://labelary.com/docs.html

Specifically:
^IL path	Image load
^IM path	Image move
^IS path, print	Image save

don, 2016-04-20:
@sclearion thanks for the links

@ivangutierrezr If the links from @sclearion don't work you might need to use the iOS and Android SDKs from [bixolon](http://www.bixolon.com/html/en/download/download_list.xhtml?all_yn=Y;). If your app is Cordova, you can write a plugin to wrap the SDK from the manufacturer.

closed 2016-04-20 by don