don/BluetoothSerial #209: How to disable the authority request dialog when enable the bluetooth

ZengYJ, 2016-03-21:
In cordova-plugin-bluetoothle ,when initializing the bluetooth, a param such as {request:false} is used to disable the tip of authority request. 
How can I get the same function like in cordova-plugin-bluetoothle?

Thank you.

don, 2016-03-24:
Can you send a screenshot of what you'd like to disable?

orenagiv, 2017-02-02:
I think he was referring to this one:
[image]

It appears if the device bluetooth is turned off.

don, 2017-02-02:
I think you'll need to edit the source and add [CBCentralManagerOptionShowPowerAlertKey](https://developer.apple.com/reference/corebluetooth/cbcentralmanageroptionshowpoweralertkey?language=objc) get rid of this dialog. [Try adding it here](https://github.com/don/BluetoothSerial/blob/3c5134339efbcea7449a498aa3b2660f779360dd/src/ios/BLE.m#L196). 

See a similar solution @H0rst has for another one of my Bluetooth plugins https://github.com/don/cordova-plugin-ble-central/issues/174#issuecomment-179775652

closed 2017-03-17 by don