don/BluetoothSerial #210: BluetoothSerial issue with ng-init

MarioArkwright, 2016-03-23:
Hello,

here is an extremely simple example Ionic app showing an issue I encounter when using BluetoothSerial plugin and ng-init directive. In the console, one can see an error message containing "ReferenceError: bluetoothSerial is not defined at Scope.$scope.isBTenabled (file:///android_asset/www/js/app.js:6:5)" when the ng-init directive is launched (i.e. when the app starts). However, one can see that there is no issue when the directive ng-click is used. 

Could you please help me about this?
Thanks in advance,

Mario 

[BTnginit.zip](https://github.com/don/BluetoothSerial/files/186370/BTnginit.zip)

don, 2016-03-24:
"bluetoothSerial is not defined" usually means that the plugin isn't installed or you're calling bluetoothSerial before Cordova's deviceready event has fired. You could try putting bluetoothSerial in a service rather than a controller. You could try looking at an older Ionic sample I have. It uses my Bluetooth Low Energy plugin but could help point you in the right direction. https://github.com/don/ionic-ble

MarioArkwright, 2016-03-25:
Thank you very much for this quick answer and for your time. I will investigate using you BLE code. Best regards, Mario

closed 2016-04-20 by don