don/BluetoothSerial #211: Connet error - HM-10

ruielias, 2016-03-28:
Hi Don,

Very good your module , but can you help me?
I want to use the HM -10 device ( CC2541 - Texas Instruments - JNHuaMao Technology Company ) where I use with your example "Chat" .
Detecting device but can not make the connection, gives error.
Let me explain what I have done from the beginning :

1º  Install SDK Android, for debuging...
2º  Install my driver Samsung Phone Android
3º  ionic start blue-test blank
4º  cd blue-test
5º  cordova plugin add cordova-plugin-bluetooth-serial
6º  cp -R \plugins\cordova-plugin-bluetooth-serial\examples\Chat\www \www 
7º  ionic platform add android
8º  ionic rum android -l -c
	The projet run in my android phone
	Detect my device
	I try connect, but report error: "Unable to connect to device"

What am I doing wrong ?

Your code chat example with my debug:

```
// (c) 2013-2015 Don Coleman
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/* global mainPage, deviceList, refreshButton, statusDiv */
/* global detailPage, resultDiv, messageInput, sendButton, disconnectButton */
/* global cordova, bluetoothSerial  */
/* jshint browser: true , devel: true*/
'use strict';

var app = {
    initialize: function() {
        this.bindEvents();
        this.showMainPage();
		console.log("Initialize...");
    },
    bindEvents: function() {
        var TOUCH_START = 'touchstart';
        if (window.navigator.msPointerEnabled) { // windows phone
            TOUCH_START = 'MSPointerDown';
        }
        document.addEventListener('deviceready', this.onDeviceReady, false);
        refreshButton.addEventListener(TOUCH_START, this.refreshDeviceList, false);
        sendButton.addEventListener(TOUCH_START, this.sendData, false);
        disconnectButton.addEventListener(TOUCH_START, this.disconnect, false);
        deviceList.addEventListener('touchstart', this.connect, false);
    },
    onDeviceReady: function() {
        app.refreshDeviceList();
		console.log("onDeviceReady...");
    },
    refreshDeviceList: function() {
        bluetoothSerial.list(app.onDeviceList, app.onError);
		console.log("Refresh Device List...");
    },
    onDeviceList: function(devices) {
        var option;
        // remove existing devices
        deviceList.innerHTML = "";
        app.setStatus("");
        devices.forEach(function(device) {
            var listItem = document.createElement('li'),
                html = '<b>' + device.name + '</b><br/>' + device.id;
			console.log("Device: " + device.name + " " +device.id);
            listItem.innerHTML = html;
            if (cordova.platformId === 'windowsphone') {
              // This is a temporary hack until I get the list tap working
              var button = document.createElement('button');
              button.innerHTML = "Connect";
              button.addEventListener('click', app.connect, false);
              button.dataset = {};
              button.dataset.deviceId = device.id;
              listItem.appendChild(button);
            } else {
              listItem.dataset.deviceId = device.id;
            }
            deviceList.appendChild(listItem);
        });
        if (devices.length === 0) {
            option = document.createElement('option');
            option.innerHTML = "No Bluetooth Devices";
            deviceList.appendChild(option);
            if (cordova.platformId === "ios") { // BLE
                app.setStatus("No Bluetooth Peripherals Discovered.");
            } else { // Android or Windows Phone
                app.setStatus("Please Pair a Bluetooth Device.");
            }
        } else {
			console.log("Found " + devices.length + " device" + (devices.length === 1 ? "." : "s."));
            app.setStatus("Found " + devices.length + " device" + (devices.length === 1 ? "." : "s."));
        }
    },
    connect: function(e) {
        var onConnect = function() {
                // subscribe for incoming data
				console.log("onConnect...");
                bluetoothSerial.subscribe('\n', app.onData, app.onError);
                resultDiv.innerHTML = "";
                app.setStatus("Connected");
                app.showDetailPage();
            };
        var deviceId = e.target.dataset.deviceId;
        if (!deviceId) { // try the parent
            deviceId = e.target.parentNode.dataset.deviceId;
        }
        bluetoothSerial.connect(deviceId, onConnect, app.onError);
    },
    onData: function(data) { // data received from Arduino
        console.log("Data: " + data);
        resultDiv.innerHTML = resultDiv.innerHTML + "Received: " + data + "<br/>";
        resultDiv.scrollTop = resultDiv.scrollHeight;
    },
    sendData: function(event) { // send data to Arduino
        var success = function() {
            console.log("success");
            resultDiv.innerHTML = resultDiv.innerHTML + "Sent: " + messageInput.value + "<br/>";
            resultDiv.scrollTop = resultDiv.scrollHeight;
        };
        var failure = function() {
			console.log("Failed writing data to Bluetooth peripheral");
            alert("Failed writing data to Bluetooth peripheral");
        };
        var data = messageInput.value;
        bluetoothSerial.write(data, success, failure);
    },
    disconnect: function(event) {
		console.log("Disconecting...");
        bluetoothSerial.disconnect(app.showMainPage, app.onError);
    },
    showMainPage: function() {
		console.log("Show Main Page...");
        mainPage.style.display = "";
        detailPage.style.display = "none";
    },
    showDetailPage: function() {
		console.log("Show Detail Page...");
        mainPage.style.display = "none";
        detailPage.style.display = "";
    },
    setStatus: function(message) {
        console.log("Status: " + message);
        window.clearTimeout(app.statusTimeout);
        statusDiv.innerHTML = message;
        statusDiv.className = 'fadein';
        // automatically clear the status with a timer
        app.statusTimeout = setTimeout(function () {
            statusDiv.className = 'fadeout';
        }, 5000);
    },
    onError: function(reason) {
		console.log("On Error reason: " + reason);
        alert("ERROR: " + reason); // real apps should use notification.alert
    }
};

`````
The output debug:

JS changed:   c:\cordova\chat\www\js\index.js
0     114767   log      Show Main Page...
1     114769   log      Initialize...
2     114933   log      Refresh Device List...
3     114934   log      onDeviceReady...
4     114937   log      Status:
5     114940   log      Device: SMC-APF 68:9E:19:10:59:2B
6     114942   log      Found 1 device.
7     114944   log      Status: Found 1 device.
8     125121   log      On Error reason: Unable to connect to device
9     130395   log      Refresh Device List...
10    130402   log      Status:
11    130407   log      Device: SMC-APF 68:9E:19:10:59:2B
12    130412   log      Found 1 device.
13    130413   log      Status: Found 1 device.
14    136819   log      On Error reason: Unable to connect to device
q
c:\cordova\chat>

**Can you help me to conect my device?**

Best regards,

Rui Elias

PriyankaRose, 2016-04-07:
hi,
have you got any solution for this problem?

ruielias, 2016-04-07:
I still I did not get ...

I will expect the DON tell me something ...

 
Date: Wed, 6 Apr 2016 22:54:55 -0700

don, 2016-04-07:
@ruielias @PriyankaRose The HM-10 is a Bluetooth Low Energy board so it won't work with Android and this plugin. The BluetoothSerial plugin uses Bluetooth Classic for Android and Bluetooth Low Energy for iOS.

Use my [Bluetooth Low Energy plugin](https://github.com/don/cordova-plugin-ble-central/) with the HM-10. Start with the  [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle).

You'll need to adjust the [UUIDs in the code](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/bluefruitle/www/js/index.js#L37-L39) to match the Bluetooth LE service on the HM-10. Check the HM-10 documentation or use a tool like [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to find the UART service on the device.

ruielias, 2016-04-09:
Hi Don,

I did as you indicated:

1. I used the NRF to know UUIDs of BLE HM-10
2. Then I used the your Bluetooth Low Energy plugin
3. I used your bluefruitle example

Result: **worked at first time.**
Thank you very much Don.

Rui Elias

closed 2016-04-20 by don