don/BluetoothSerial #213: ionic and bluetoothSerial works but not in factory

AndrewListerHK, 2016-04-05:
I've been banging my head against this problem for too long and finally caved in and asking the experts....  I am a relative noob to all of this stuff.

Based on the SimpleSerial example, I am **successfully** making a call when ionic is ready:

```
        $ionicPlatform.ready(function() {
               .
               .
                bt.isEnabled();
        }
```
 to a javascript file containing
```
var bt = {
    enabled: false,
    devices: [],
    error: "",

    isEnabled: function() {
        console.log("isEnabled called");
        bluetoothSerial.isEnabled(
            function() {
                alert("Bluetooth is enabled");
                return true;
            },
            function() {
                alert("Bluetooth is NOT enabled");
                return false;
            }
        );
    },
```

With my limited knowledge, it does not seem as though these direct calls from the UI are possible without a controller and then a factory (I think).  

Controller:
```
    .controller('BluetoothCtrl', function ($cordovaBluetoothSerial, $scope, Bluetooth) {
        $scope.isEnabled = Bluetooth.isEnabled();
    }
```
Factory:
```
angular.module('glorify.bluetooth', ['ionic', 'ngCordova', 'ngCordova.plugins.bluetoothSerial'])

    .factory('Bluetooth', function($cordovaBluetoothSerial) {
        console.log("Bluetooth service");

        return {
            isEnabled: function() {
                return bt.isEnabled();
            },
        }
    });
```

but I immediately get these errors on the android device.
```
I/chromium(21628): [INFO:CONSOLE(25642)] **"ReferenceError: bluetoothSerial is not defined**
I/chromium(21628):     at Object.bt.getDevices (file:///android_asset/www/js/bluetooth.js:19:9)
I/chromium(21628):     at Object.<anonymous> (file:///android_asset/www/js/services/bluetooth.js:14:26)
I/chromium(21628):     at Object.invoke (file:///android_asset/www/lib/ionic/js/ionic.bundle.js:17762:17)
I/chromium(21628):     at Object.enforcedReturnValue [as $get] (file:///android_asset/www/lib/ionic/js/ionic.bundle.js:17615:37)
I/chromium(21628):     at Object.invoke (file:///android_asset/www/lib/ionic/js/ionic.bundle.js:17762:17)
```

I have tried injecting everything I can think of into the factory but must be missing something very obvious or I've done it wrongly in the first place.  

Also, I can see that there is already a factory defined in `ng-cordova.js` - is there an easy way to use this?

Please help!!

Thanks in advance.

Andrew

don, 2016-04-08:
You need to make sure bluetoothSerial isn't called before Cordova deviceready (or the Ionic equivalent fires.)

I have an older example of using Ionic with my [Bluetooth Low Energy plugin](https://github.com/don/cordova-plugin-ble-central). It might help you solve your Ionic issues with this plugin.

For my NFC plugin I needed to wrap calls to the plugin in the factory with `$ionicPlatform.ready`
https://github.com/don/ionic-nfc-reader/blob/master/www/js/app.js#L30-L43

AndrewListerHK, 2016-04-08:
Hi Don

Thanks for your reply.  As I said, I am quite new to all of this ionic, angular and cordova stuff but quite an experienced programmer in more “normal” languages ;-)

I did have some success after I posted achieved by waiting for the device to be ready, as you stated.  As far as the angular injection idea, I still haven’t quite figured out if there should be any particular order…?

A slight problem I have now is that the bluetoothSerial is only capable of connecting to a single device.  I am writing an IoT app that is intended to connect to multiple devices and at the moment, our PCB only has an BT2.0 SPP module on board - this will change to BLE soon).  I was wondering how difficult it would be to change your module (putting hand up) to use the address as an index for subscribe, etc calls.  Disclaimer: I have not looked at any of the code yet!  I did read of some having success by instantiating several instances but I do not know how many devices I will have.

Thanks for your outstanding work on these cordova modules BTW.

Cheers

Andrew

don, 2016-04-08:
Previously you asked how the ng-cordova factory works. I haven't used it, so I'm not sure.

When you're injecting the objects into the angular module, order doesn't matter.

People have asked for multi-device support. It would be a nice feature, but I don't have any near term plans to add it. (I'm doing mostly BLE work instead of classic.) Last time I looked it was going to be difficult to add without breaking the API. I really don't want to break the API for existing users. I'm willing to consider pull requests or talk about a project to enhance it.

AndrewListerHK, 2016-04-08:
Thanks for the info on angular order.  I should have done some tutorials on a lot of this stuff but chose to dive right in and bang my head on many walls.

For multi device, I would not break the existing API - but I’ll have a look at the code before making any decisions - keeping you informed.

Cheers

Andrew

Sundypha, 2016-05-06:
In ng-cordova the plug-ins are already in a factory. You have to use their way of calling the functions `$cordovaPlugin.functionToCall().then(successCallback, failCallback);` make sure to include ng-cordova.js after ionic/angular.js and before cordova.js because ng-cordova.js depends on angular.js (for obvious reasons).

closed 2017-03-17 by don