don/BluetoothSerial #214: bluetoothSerial.isEnabled not working on my Android device

MarijnBrosens, 2016-04-07:
Hi Don, 

I added the BluetoothSerial plugin to my project ( a simple phonegap project ).
I try to connect my Android ( onePlus 2 ) device to an arduino via hc-06.
I added the following code to my index.js to test but i can't get alerts at all.

index.js

```
var app = {
    initialize: function() {
        this.bindEvents();
    },
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    onDeviceReady: function() {
        bluetoothSerial.isEnabled(
            function() { 
                alert("Bluetooth is enabled");
            },
            function() { 
                alert("Bluetooth is *not* enabled");
            }
        );
        app.receivedEvent('deviceready');
    },
    // Update DOM on a Received Event
    receivedEvent: function(id) {
        var parentElement = document.getElementById(id);
        var listeningElement = parentElement.querySelector('.listening');
        var receivedElement = parentElement.querySelector('.received');

        listeningElement.setAttribute('style', 'display:none;');
        receivedElement.setAttribute('style', 'display:block;');

        console.log('Received Event: ' + id);
    }
};

app.initialize()`

```

config.xml

```

<plugin name="cordova-plugin-bluetooth-serial" spec="0.4.5" source="rgb" />

```

-> is this even possible?

With your BLE plugin i did the same, this part worked, but i couldn't scan my hc-06.

```

ble.isEnabled(
    function() {
        alert("Bluetooth is enabled ble");
    },
   function() {
        alert("Bluetooth is *not* enabled ble");
   }
);

```

don, 2016-04-08:
The HC-06 is Bluetooth Classic right? If so you'll need to use this plugin and not the [BLE one](https://github.com/don/cordova-plugin-ble-central)

Did you manually add that line to config.xml? You shouldn't need to.

Your code works OK. Here's how I tested it.

$ cd /tmp
$ cordova create issue214
$ cd issue214
$ cordova platform add android
$ cordova plugin add cordova-plugin-bluetooth-serial
_add your code to www/js/index.js_
$ cordova run android --device && adb logcat

I see the dialogs on my phone.

If you're phone is running Android 6, you might need to edit config.xml and add target version 22 until I fix #194. 

    <platform name="android">
        <allow-intent href="market:*" />
        <preference name="android-targetSdkVersion" value="22"/> 
    </platform>

c7devteam, 2016-04-22:
I'm having the same problem like Marjin.

I've added
`<preference name="android-targetSdkVersion" value="22"/> `
to the config.xml. But I'm using Android 5.1.1. 

It looks like it's hanging forever in bluetoothSerial.isEnabled. I've binded a bluetooth device to my android tablet and BT is enabled.

`/*
 * Licensed to the Apache Software Foundation (ASF) under one
 * or more contributor license agreements.  See the NOTICE file
 * distributed with this work for additional information
 * regarding copyright ownership.  The ASF licenses this file
 * to you under the Apache License, Version 2.0 (the
 * "License"); you may not use this file except in compliance
 * with the License.  You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing,
 * software distributed under the License is distributed on an
 * "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
 * KIND, either express or implied.  See the License for the
 * specific language governing permissions and limitations
 * under the License.
 */
var app = {
    // Application Constructor
    initialize: function() {
        this.bindEvents();
    },
    // Bind Event Listeners
    //
    // Bind any events that are required on startup. Common events are:
    // 'load', 'deviceready', 'offline', and 'online'.
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    // deviceready Event Handler
    //
    // The scope of 'this' is the event. In order to call the 'receivedEvent'
    // function, we must explicitly call 'app.receivedEvent(...);'
    onDeviceReady: function() {
        app.receivedEvent('deviceready');

        var button_connect = document.getElementById("button_connect");
        button_connect.addEventListener('touchend', app.bluetoothConnect);

       /*bluetoothSerial.enable(
        function() {
            app.log("Bluetooth is enabled");
        },
        function() {
            app.log("The user did *not* enable Bluetooth");
        });*/

        console.log("calling isEnabled?")
        bluetoothSerial.isEnabled(
           app.bluetoothlistPorts,
           app.bluetoothNotEnabled
       );

    },
    // Update DOM on a Received Event
    receivedEvent: function(id) {
        var parentElement = document.getElementById(id);
        var listeningElement = parentElement.querySelector('.listening');
        var receivedElement = parentElement.querySelector('.received');

        listeningElement.setAttribute('style', 'display:none;');
        receivedElement.setAttribute('style', 'display:block;');

        app.log('Received Event: ' + id);
    },

    bluetoothConnect: function(){
        app.log("yeah");
    },

    log: function(text){
      var log_area = document.getElementById("log_area");
      log_area.value = text + "\n" + log_area.value;
    },

    bluetoothlistPorts: function() {
      console.log("WTF");
        // list the available BT ports:
        bluetoothSerial.list(
            function(results) {
                app.log("----------------------")
                app.log(JSON.stringify(results));
            },
            function(error) {
                app.log("----------------------")
                app.log(JSON.stringify(error));
            }
        );
    },

    bluetoothNotEnabled: function() {
        console.log("Bluetooth is not enabled.");
    }
};

`

closed 2017-03-17 by don