don/BluetoothSerial #216: The correct way to connect multi device

hypery11, 2016-04-08:
I've read the previous same[ issue](https://github.com/don/BluetoothSerial/issues/58), there are many solutions but can not work actually,
is there any right way to connect multi device?

don, 2016-04-08:
@hypery11 right now there's not a good way to connect to multiple devices. 

/cc @AndrewListerHK

Shikoruma, 2016-04-12:
My fork should work on Android thanks to @hypery11
plugin https://github.com/Shikoruma/BluetoothSerial
test app on ionic https://github.com/Shikoruma/blueserialtest

AndrewListerHK, 2016-04-12:
That's great news Shikoruma.  I'll try it out and let you know.

andrewinHK, 2016-04-14:
What do you think about making the mac address the first argument in all the functions?  It would kind of make it a little more standard as Don's original code used bluetoothSerial.connect(macAddress_or_uuid, connectSuccess, connectFailure);

Shikoruma, 2016-04-14:
I thought of it but I wanted this version backward compatible with the current version.
I also think that most user will not use the multi-connection.

andrewinHK, 2016-04-14:
Ah yes - of course.

closed 2017-03-17 by don

lovelyelfpop, 2020-04-12:
Try this one https://github.com/bugnano/cordova-plugin-networking-bluetooth