don/BluetoothSerial #218: bluetoothSerial is not defined

diattamo, 2016-04-11:
Hello,

I am trying to access the bluetoothSerial methods following basically the logic in the examples. However I am getting an error saying "bluetooth serial is not defined" and I need help figuring it out. I am probably missing something considering that I am not really familiar with angularjs. 

Here is what I am trying to do. The weigh function is what I am calling in my html page and I have made sure to add the plugin in the config file. 

$scope.weigh = function() {
        console.log('weighing');
   
	   	bluetoothSerial.connect("98:D3:31:30:76:9E", $scope.onConnect, $scope.onDisconnect);
	    //$scope.bluetoothSerial.write("1");

	    };
	   

	$scope.onConnect = function() {
		    //bluetoothSerial.subscribe("\n", weigh.onMessage, weigh.subscribeFailed);
		    console.log('Connected');
		    $scope.statusDiv="Connected to " + "98:D3:31:30:76:9E" + ".";        
		};

	$scope.onDisconnect = function() {
		    //$window.alert("Disconnected");
		    console.log('Disconnected');
		    $scope.statusDiv="Disconnected.";
		};
	$scope.onMessage = function(data) {
			console.log('Received data');
		    $scope.sensorData = data;
		};
	$scope.subscribeFailed = function() {
		    //$window.alert("subscribe failed");
		    console.log('subscribe failed');
		    $scope.statusDiv="Disconnected.";

		};

AndrewListerHK, 2016-04-12:
Please see the recent discussion here: https://github.com/don/BluetoothSerial/issues/213

Are you calling the function on a handheld device?  You mention calling it on an html page so I'm not sure.  The plugin accesses a bluetooth module on a phone etc and will not work on a standard browser.

don, 2016-04-20:
Get a simple Bluetooth example working before adding Angular. 

Run on an Android device, it won't work in the emulator or browser. 
See https://github.com/don/BluetoothSerial/issues/213#issuecomment-207476819

closed 2016-04-20 by diattamo