don/BluetoothSerial #219: Phonegap - Plugin for Bluetooth Printer?

Divyaadz, 2016-04-15:
Thanks for great effort is creating this plugin.

I have following questions
1. Whether this plugin support to print receipt in Bluetooth Printer?

2. Whether bluetoothSerial.write method accept HTML string as input parameter and it get printed in printer with designed format?
    Eg: bluetoothSerial.write("<table><tr><td>hello, world<tr><td></table>", success, failure);

3. Can I get a complete phonegap example for this plugin? like getting mac address and writing the sample content to bluetooth device.

don, 2016-04-20:
@Divyaadz `bluetoothSerial.write` should accept HTML (but I think it only access ASCII characters). Alternately you can pass binary data using ArrayBuffer. See [ble.write](https://github.com/don/BluetoothSerial#write)

I'm not sure about writing to your printer. From Android, it could work if your printer accepts Serial Port Profile. You'd need to see if the printer can accept plain text or needs something like postscript or pcl.

Check out the [example directory](https://github.com/don/BluetoothSerial/tree/master/examples). You should be able to connect the [Chat example](https://github.com/don/BluetoothSerial/tree/master/examples/Chat) to a [bluetooth serial terminal](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo?utm_source=chrome-ntp-launcher) on your computer.

closed 2016-04-20 by don

don, 2016-04-20:
See #207 for more info

bottasergio, 2017-07-13:
@Divyaadz  did you solve the problem? I am trying to print using this plugin to a Citizen CMP-30 printer...
I am replacing the **data** parameter with a string like this:

`			data = CHAR_ESC+ "%-12345X@PJL JOB NAME = \"18-01\" <CR><LF>\n" +
                        "@PJL ENTER LANGUAGE = CPCL <CR><LF>\n" +
								CHAR_ESC + "EPor fin imprimiste LRPMQTP!" +
                        CHAR_ESC + "cAPor fin imprimiste LRPMQTP!" +
                        CHAR_ESC + "%-12345X @PJL EOF <CR><LF>\n" +
                        CHAR_ESC + "%-12345X";	`

where CHAR_ESC is `var CHAR_ESC = 0x1B;			`

but the printer only move one blank line up and doesn't print.

magnus223, 2017-07-27:
@bottasergio I am working on a bluetooth printer as well and my printer prints and responds to commands correctly when commands are send in unicode format. ex. 
```
var LF = "\n" ;
var ESC ='\u001b';
```

bottasergio, 2017-07-27:
It´s working for me with this code (sending CPCL commands to printer):

`var CHAR_ESC = 0x1b;
var LINE_FEED = '\n'; 
var CARRIAGE_RETURN = '\r';`

`var data = "! 0 200 200 500 1" + CARRIAGE_RETURN + LINE_FEED +`
`"TEXT 4 0 0 45 sometext" + CARRIAGE_RETURN + LINE_FEED +`
`"TEXT 4 0 155 45 Lorem ipsum" + CARRIAGE_RETURN + LINE_FEED +`
`"INVERSE-LINE 0 45 145 45 45" + CARRIAGE_RETURN + LINE_FEED +`
`"PRINT" + CARRIAGE_RETURN + LINE_FEED ;`
`bluetoothSerial.write(data, function(){alert('ok');}, function(){alert(Error!');});`

yekogarcia, 2019-11-05:
As I do to bold the letter and give a larger size to the header neither does it recognize the tildes with Bluetooth Serial Plugin for PhoneGap