don/BluetoothSerial #220: Data not received in other end device.

PriyankaRose, 2016-04-16:
hi Don,

Greetings.
I have tired your BluetoothSerial Chat plugin to transfer set of data from one device to another (android-android and ios-ios).
while testing for android-android, it lists the devices and both are even connected. From device-1 if i send data, it says success. but it is not received in the device-2!

for your reference, i am attaching the screens took while testing in both android devices
Device-1
[image]
[image]
[image]
[image]
[image]

Device-2
[image]
[image]

and here is my index.js file
`var app = {
    initialize: function() {
        this.bindEvents();
        this.showMainPage();
    },
    bindEvents: function() {

        var TOUCH_START = 'touchstart';
        if (window.navigator.msPointerEnabled) { // windows phone
            TOUCH_START = 'MSPointerDown';
        }
        document.addEventListener('deviceready', this.onDeviceReady, false);
        refreshButton.addEventListener(TOUCH_START, this.refreshDeviceList, false);
        sendButton.addEventListener(TOUCH_START, this.sendData, false);
        disconnectButton.addEventListener(TOUCH_START, this.disconnect, false);
        deviceList.addEventListener('touchstart', this.connect, false);
    },
    onDeviceReady: function() {
        app.refreshDeviceList();
    },
    refreshDeviceList: function() {
        bluetoothSerial.list(app.onDeviceList, app.onError);
		app.onData();
    },
    onDeviceList: function(devices) {
        var option;

        // remove existing devices
        deviceList.innerHTML = "";
        app.setStatus("");

        devices.forEach(function(device) {

            var listItem = document.createElement('li'),
                html = '<b>' + device.name + '</b><br/>' + device.id;

            listItem.innerHTML = html;

            if (cordova.platformId === 'windowsphone') {
              // This is a temporary hack until I get the list tap working
              var button = document.createElement('button');
              button.innerHTML = "Connect";
              button.addEventListener('click', app.connect, false);
              button.dataset = {};
              button.dataset.deviceId = device.id;
              listItem.appendChild(button);
            } else {
              listItem.dataset.deviceId = device.id;
            }
            deviceList.appendChild(listItem);
        });

        if (devices.length === 0) {

            option = document.createElement('option');
            option.innerHTML = "No Bluetooth Devices";
            deviceList.appendChild(option);

            if (cordova.platformId === "ios") { // BLE
                app.setStatus("No Bluetooth Peripherals Discovered.");
            } else { // Android or Windows Phone
                app.setStatus("Please Pair a Bluetooth Device.");
            }

        } else {
            app.setStatus("Found " + devices.length + " device" + (devices.length === 1 ? "." : "s."));
        }

    },
    connect: function(e) { alert('1');
        var onConnect = function() {
                // subscribe for incoming data
                bluetoothSerial.subscribe('\n', app.onData, app.onError);

                resultDiv.innerHTML = "";
                app.setStatus("Connected");
                app.showDetailPage();
            };

        var deviceId = e.target.dataset.deviceId;
        if (!deviceId) { // try the parent
            deviceId = e.target.parentNode.dataset.deviceId;
        }

        bluetoothSerial.connect(deviceId, onConnect, app.onError);
    },
    onData: function(data) { // data received from Arduino
        alert("data="+data);
        resultDiv.innerHTML = resultDiv.innerHTML + "Received: " + data + "<br/>";
        resultDiv.scrollTop = resultDiv.scrollHeight;
    },
    sendData: function(event) { // send data to Arduino
	//connect();
		
        var success = function() {
            alert("success");
            resultDiv.innerHTML = resultDiv.innerHTML + "Sent: " + messageInput.value + "<br/>";
            resultDiv.scrollTop = resultDiv.scrollHeight;
        };

        var failure = function() {
            alert("Failed writing data to Bluetooth peripheral");
        };

        var data = messageInput.value;
        bluetoothSerial.write(data, success, failure);
    },
    disconnect: function(event) {
        bluetoothSerial.disconnect(app.showMainPage, app.onError);
    },
    showMainPage: function() {
        mainPage.style.display = "";
        detailPage.style.display = "none";
    },
    showDetailPage: function() {
        mainPage.style.display = "none";
        detailPage.style.display = "";
    },
    setStatus: function(message) {
        alert("message="+message);

        window.clearTimeout(app.statusTimeout);
        statusDiv.innerHTML = message;
        statusDiv.className = 'fadein';

        // automatically clear the status with a timer
        app.statusTimeout = setTimeout(function () {
            statusDiv.className = 'fadeout';
        }, 5000);
    },
    onError: function(reason) {
        alert("ERROR: " + reason); // real apps should use notification.alert
    }
};`

my problem is, data not received in the device-2.

Thanks in advance. :)

PriyankaRose, 2016-04-19:
the devices tested are  
device-1 : Moto G3
device-2 : Lenovo Vibe K4 Note

and i have tested with other devices too.
Data Read not working though the Data sent success message comes.

Could you provide any suggestion.!!

don, 2016-04-20:
@PriyankaRose Currently can't connect Android to Android. You can pair two Android devices, but unless one of them opens provides Serial Port Profile, you're not going to be able to connect. This plugin doesn't currently support incoming connections. Some of the other github issues may point to forks from other users that fix this. https://github.com/don/BluetoothSerial/issues/50#issuecomment-66405396 It's technically possible, I just haven't implemented it and no one has submitted a pull request.

Android to iOS is not possible with this plugin. Android uses Bluetooth Classic and iOS uses Bluetooth Low Energy.

You could do iOS to iOS (or Android) is the other device implements one of the UART-like BLE SPP services. https://github.com/don/BluetoothSerial/issues/75#issuecomment-52591397 You'd need to implement the BLE service with native code. I don't think there's a Corodva plugin for that yet.

closed 2016-04-20 by don