don/BluetoothSerial #222: How to read raw data?

shiefzhang, 2016-04-25:
I am using Ionic to connect the bluetooth serial.
In my case, the subscribeRawData() is not work, so I have to use read() to read data.
It worked, but the char code that large than 0x7F was converted to 0xFD.

Is there the readRawData() function to read the raw data?

shiefzhang, 2016-04-28:
I try to add a new function named "readRaw()" to the plugin.
In www/bluetoothSerial.js, I added:
`
  // read all the raw data in the buffer
  readRaw: function (success, failure) {
    cordova.exec(success, failure, "BluetoothSerial", "readRaw", []);
  },
`
In src/android/com/megster/cordova/BluetoothSerial.java, I added:
`
  private static final String READ_RAW = "readRaw";
===================================================
    } else if (action.equals(READ_RAW)) {

      callbackContext.success(readRaw());
===================================================
  private String readRaw() {
    int length = buffer.length();
    String data = buffer.substring(0, length) + " test";
    buffer.delete(0, length);
    return data;
  }
`

When I use “$cordovaBluetoothSerial.readRaw();” to invoke the function,
I get the error message:"readRaw() is not a function".

Does I miss something in the code?
Anyone can give some suggest?

don, 2016-04-28:
Currently there's no method to read raw data. If you can get the data through `read` you should be able to get the data from `subscribeRawData`.

As data arrives from the peripheral, a String version and the raw versions are processed
https://github.com/don/BluetoothSerial/blob/master/src/android/com/megster/cordova/BluetoothSerialService.java#L450-L463

String data is saved in a buffer and send to subscribers, raw data is only sent to subscribers. 
https://github.com/don/BluetoothSerial/blob/master/src/android/com/megster/cordova/BluetoothSerial.java#L350-L363

Since the raw data is not saved, it's going to be more difficult for you to implement read_raw.

Especially with Bluetooth Classic, I think that `subscribeRawData` is a better solution than `readRaw`. You can always buffer the data in your Javascript code if you want readRaw.

Check you original code again. Remember that the onData callback will get an ArrayBuffer and you'll need to turn this into a typed array to access the data.

    var onData =  function(buffer) {
        console.log(new Uint8Array(buffer));
    }

    bluetoothSerial.subscribeRawData(onData, onError);

toregua, 2016-04-29:
Hello,

I am exactly in the same context as shiefzhang with Ionic and BluetoothSerial and the subscribeRawData success callback is never call (failure too).

`       $cordovaBluetoothSerial.write(data).then(
            function (success) {
                $cordovaBluetoothSerial.available().then(
                function (numBytes) {
                    console.log("There are " + numBytes + " available to read.");
                    $cordovaBluetoothSerial.subscribeRawData().then(
                        function (data) {
                          var temp= data;
...
                        },
                        function (error) {
                            alert(JSON.stringify(error));
                        }
                 },
                function (error) {
                    alert(JSON.stringify(error));
                }
               );
            },
            function (err) {
                $scope.showAlert = function () {
                    var alertPopup = $ionicPopup.alert({
                        title: 'Write Error : ',
                        template: JSON.stringify(err)
                    });
                    alertPopup.then(function (res) {
                    });
                };
            }
        );

When I use the Read function I have result i think in ansi but i don't know how to convert it in unicode to be human readable.

Do you have any ideas please ?

shiefzhang, 2016-04-29:
@don 
thanks for responses.

I used the Ionic framework that base on angularjs and bootstrap.
And used ngCordova plugin to connect the cordova plugin to angularjs.
Maybe something wrong on my coding, any subscribe function is not working.
So I have to use the read() function.

shiefzhang, 2016-04-29:
@toregua 

Yes, maybe we have same problem.

Now, I am trying to rewrite the read() function in src/android/com/megster/cordova/BluetoothSerial.java like this:
```
  private byte[] readRaw() {
    int length = buffer.length();
    String data = buffer.substring(0, length);
    **rawData=new String(data.getBytes(),"UTF-8");**
    buffer.delete(0, length);
    return rawData;
  }
```
but I got a UnsupportedEncodingException error, but I think this is a right way.

toregua, 2016-04-29:
Finaly I have successfully call the subscribeRawData method like that : 

```
    $scope.Test = function () {

        var data = new Uint8Array(6);
        data[0] = 0x00;
        data[1] = 0x06;
        data[2] = 0xFD;
        data[3] = 0x00;
        data[4] = 0xA0;
        data[5] = 0x91;

        bluetoothSerial.subscribeRawData(function (data) {
            var bytes = new Uint8Array(data);
            if (bytes.length == 25) {
                var hexa = utilService.Uint8ArrayToHex(bytes);
                console.log(hexa);
            }
            console.log(bytes);
        }, function (error) { alert(JSON.stringify(error)); });

        $cordovaBluetoothSerial.clear().then(
            function (success) {
                $cordovaBluetoothSerial.write(data).then(
                    function (success) {

                    },
                    function (err) {
                        $scope.showAlert = function () {
                            var alertPopup = $ionicPopup.alert({
                                title: 'Erreur Write : ',
                                template: JSON.stringify(err)
                            });
                            alertPopup.then(function (res) {
                            });
                        };
                    }
                );
            },
            function (err) {
                var test = err;
            }
        );
    };
```

shiefzhang, 2016-05-03:
@toregua 

You are a genius!
It worked!

Using ngCordova plugin, functions must be added a prefix, such as:
`
bluetoothSerial.clear()-->$cordovaBluetoothSerial.clear()
`

But the callback function must be the same as original, such as:
`
bluetoothSerial.subscribeRawData()-->bluetoothSerial.subscribeRawData()
`

It's  amazing.

toregua, 2016-05-03:
@shiefzhang happy to have helped you ;) it is true that this behavior is strange...but now we have a workaround

don, 2016-05-03:
Glad you got it working. The ng wrapper is not my code so I'm not sure what's happening there. My code has some API calls where the success function is called many times (scan, subscribe, subscribeRawData.) Promises don't work well for those calls.

closed 2016-05-03 by don

RavikumarRapolu, 2017-10-14:
Hai iam also trying to write some data to printer, but iam unable to write data. but iam getting success callback and OK status in call back. Can you please share working code to print data