don/BluetoothSerial #223: read from two HC-06s

diattamo, 2016-04-25:
Hello I am trying to read data from two HC-06 modules each connected to their own arduino. 
What I am trying to do is:
connect to 1, disconnect from it, connect to the second one and and then disconnect again. I have gotten it to work once but I keep running into this issue. 

> 04-25 11:34:44.703 22437-23021/com.phonegap.helloworld E/BluetoothSerialService: java.io.IOException: read failed, socket might closed or timeout, read ret: -1

Sometimes it connects, but only to the first module. 

Here is my code (below). Any suggestions as to how I should be handling the flow of connect to 1->disconnect->connect to 2->disconnect will be appreciated.

`var app = {
    initialize: function() {
        this.bindEvents();
        localStorage.currentUnit = JSON.stringify("kg");
    },
    initializeSettings: function() {
        document.getElementById("current-unit").innerHTML = JSON.parse(localStorage.currentUnit);
    },
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
        Measure.addEventListener('touchstart', this.connect1, false);
    },
    onDeviceReady: function() {
         console.log("device ready");
    },

    connect1: function() {
        console.log("connecting to bluetooth 1");

        var onConnect = function() {
                // subscribe for incoming data
                bluetoothSerial.subscribe("\n", app.onMessage1, app.subscribeFailed); 
                bluetoothSerial.write("1");
            };

        bluetoothSerial.connect(macAddress, onConnect, app.onFailconnect);

    },
    connect2: function() {
        console.log("connecting to bluetooth 2");

        var onConnect = function() {
                // subscribe for incoming data
                bluetoothSerial.subscribe("\n", app.onMessage2, app.subscribeFailed); 
                bluetoothSerial.write("1");
            };

        bluetoothSerial.connect(macAddress2, onConnect, app.onFailconnect);
    },
    onFailconnect: function() {
        console.log("connection failed");
    },
    onMessage1: function(data) {
        console.log("reading 1");
        counter.innerHTML = data;
        bluetoothSerial.disconnect(app.connect2, app.failDisc);

    },
    onMessage2: function(data) {
        console.log("reading 2");
        counter.innerHTML = "data2";
        bluetoothSerial.disconnect(app.successDisc, app.failDisc);
    },
    subscribeFailed: function() {
        alert("subscribe failed");
    },
    successDisc: function() {
        console.log("Disconnect successful");
    },
    failDisc: function() {
        console.log("Disconnect Unsuccessful");
    },

};`

don, 2016-04-28:
This should work. If you disconnect and reconnect to the **same** device do you get this error?  (Try with each device.)

After you disconnect from the peripheral, is there something that need to be reset before it will accept a new connection?

Is it a timing issue? If you wait a bit, can you successfully connect?

diattamo, 2016-04-30:
This actually worked a bit for awhile this to work a few times but very inconsistently. Unlike what I had previously thought, the issue might actually have been with the circuitry for the bluetooth modules. 

Hopefully fixing that is going to solve our problem.

Thank you for your reply.

closed 2016-04-30 by diattamo