don/BluetoothSerial #225: Devices with more than one serial port (SPP)?

kiwisoup, 2016-05-02:
Hi,

I'm attempting to connect to a device that presents multiple serial ports.  I can connect to the device but I get no response to any of the device commands. I suspect that I am getting a connection to the wrong serial port.

To test the device I connected via a terminal in windows.  When connecting to the first comm port (by number order) I get the same result as when using the plugin with my Android device.  When I use the second comm port (by number order) I get the correct response to the issued commands.

Is there are way to select the Comm port?

thank you
Steve.

don, 2016-05-03:
@kiwisoup That's a good question. I'm not sure. The code is connecting here https://github.com/don/BluetoothSerial/blob/master/src/android/com/megster/cordova/BluetoothSerialService.java#L351 Maybe your device is using different UUIDs for each port? Can you step through in the Android debugger to see what's happening?

kiwisoup, 2016-05-05:
Thanks for info (and the entire piece of work) Don.  
I'll brush up on my Android debugger skills and get back with more info soon.

don, 2016-05-05:
Try opening your project's platforms/android directory with [Android Studio](http://developer.android.com/tools/studio/index.html) to debug.

kiwisoup, 2016-05-09:
Hi Don,
I've got a demo application from the device vendor and indeed it does use a different UUID.  In my initial testing it didn't seem to make much difference (I still couldn't get retuned data).  I'm about to debug the demo application to see exactly what commands are sent to the device to check for a triggering sequence,  I'll report back when I have something useful to add.

kiwisoup, 2016-05-10:
Using the different UUID did the trick.  There's no obvious sign that the connection isn't working with the standard UUID, just no data returned.  Once the connection was made with the UUID used by the device vendor, data was returned as expected.  In my previous comment I mentioned that I have no data returned after changing the UUID.  There was an error in the my callback function! 

Thank you very much for your help, and again for the plugin as a whole.  It would have taken me quite some time to find the UUID without your guidance.

closed 2016-05-10 by kiwisoup