don/BluetoothSerial #226: Unpaired discovery always return [] object

mumairali, 2016-05-05:
Hi Don,

I am trying to reterive unpaired devices list but each time it returns empty object each time. Following is code that I have tried:

```
          bluetoothSerial.discoverUnpaired(
                function(result) {
                    app.display(JSON.stringify(result));
                },
                function(error) {
                    app.display("Problem while discovering");
                }
            );
```

it work fine if I listed out paired devices. 
NOTE: my phone search show me unpaired devices list but I am wired why above code is not working.

Please help me what I am doing wrong!

don, 2016-05-05:
This might be an Android permission problem.  Until I fix it, can you edit config.xml and target Android 22

    <preference name="android-targetSdkVersion" value="22"/>

Sundypha, 2016-05-05:
I tried this evening to get a quick fix. It's very dirty! maybe on the weekend I will create a fork and make it pretty for a PR.
Try to replace your BluetoothSerial.java with this gist: https://gist.github.com/Sundypha/9259044aeb2d4e17929989adba4a1b79

It works on my device but I don't know if I break older sdkVersions.

Sundypha, 2016-05-06:
see #194

mumairali, 2016-05-06:
thanks @Sundypha for quick help..

I am getting error on line number:224, 252 in android studio
```
cordova.requestPermissions <- can't resolve
cordova.hasPermission <- can't resolve
```

That mean I should wait for fix. I hope that will be available very soon :-)

thank you @don

Sundypha, 2016-05-06:
@mumairali try to update the platform `cordova platform update android` maybe this updates the cordova files. I use this in my ionic app with ng-cordova.js.  Currently I'm traveling. I could help you tomorrow.  I totally forgot about the manifest.xml you can add it in the plugin.xml but you already figured that out.

L2Gui, 2016-06-14:
Same issue here :-/

commit 408a03a0 (2016-09-17): Add support for Android 6.0 (API 23) permissions

Fixes #194, Fixes #226
This is a different implementation of @Sundypha's pull request #228
closed 2016-09-17 by don