don/BluetoothSerial #227: ionic

rakesanug, 2016-05-05:
Hi Don,

      i build the bluetooth chat app using ionic. and i am successfully connect with two phone. i cant write and read messages can u help me
i using this plugin:
   cordova plugin add cordova-plugin-bluetooth-serial
the following codes are i am writing

$scope.connectDevice=function()
  {
       $cordovaBluetoothSerial.connectInsecure(localStorage.address).then(function(result){
       console.log('connected');
       console.log(result)
       },
       function(er)
       {
           console.log('not connected');
           console.log(er);
       })

  }
  $scope.listOfDevices=function()
  {
      $cordovaBluetoothSerial.list().then(function(result)
      {
      console.log(result);
      $scope.chats=result;
//            $scope.btlist=result;

     },
      function(error){
       console.log(error)   ;
      });

  }
  $scope.readDevice=function()
  {
      console.log('read device');
      console.log($cordovaBluetoothSerial)
      //$cordovaBluetoothSerial.setDiscoverable(0);
      $cordovaBluetoothSerial.setDeviceDiscoveredListener(function(device) {
          console.log('Found: '+device.id);
      });
      $cordovaBluetoothSerial.subscribe('\n', function (data) {
          console.log(data);
      }, function(obj)
      {
        console.log(obj)
      });
      $cordovaBluetoothSerial.subscribeRawData(function (data) {
          var bytes = new Uint8Array(data);
          console.log(bytes);
      }, function(obj)
      {
          console.log(obj)
      });
      $cordovaBluetoothSerial.available().then(function (numBytes)
      {
          console.log("There are " + numBytes + " available to read.");
          $cordovaBluetoothSerial.readUntil('\n').then(function (data) {
                    console.log(data);
                }, function(er){
                    console.log(er);
                });

      }, function(r)
      {
          console.log(r)
      });

  }
$scope.writeDevice=function()
  {
         console.log('write device');
          $cordovaBluetoothSerial.write("hello world").then(function(re)
             {
                 console.log('successfully write')
                 console.log(re)
             },
             function(e)
             {
                 console.log(e)
             });

  }

don, 2016-05-05:
Currently plugin doesn't work between two phones. You'll need other software to open the RFCOMM socket on one of the phones. This plugin will connect to the socket (on Android.)

Some people are working on forks of this plugin will that support incoming connections, but there's not definitive solution yet. See https://github.com/don/BluetoothSerial/issues/50#issuecomment-66405396

rakesanug, 2016-05-06:
> connections, but there's not definitive solution yet. See #50 (comment)

hi,

    Thank you for your reply. i  try as you suggested. and let you know
once i fixed the code.

Thanks & Regards

Gunasekar.C

rakesanug, 2016-05-09:
hi,
  i just tried the plugin what you suggested its working in paired devices intex/samsung galaxy tab s2. but other devices(asus and galaxy tab s2) is not working ,

the following plugins are used

cordova plugin add com.megster.cordova.bluetoothserial

don, 2016-05-09:
Can you try with the latest version of the plugin? The plugin id has been changed

    cordova plugin rm com.megster.cordova.bluetoothserial
    cordova plugin add cordova-plugin-bluetooth-serial

closed 2016-09-17 by don