don/BluetoothSerial #229: Return device id from list or discoverUnpaired

mumairali, 2016-05-09:
Hello Don,

I am trying to retrieve the `device.id` and then `connect()` to the `device.id`. but I am getting `undefined`. Here is my code,

```
_searchPaired : function(name) {
        bluetoothSerial.list(
            function(devices) {
                devices.forEach(function(device) {
                    if (device.name === name) {
                        // found in paired
                        return device.id;

                    } else {
                        return "";
                    }   // else
                });
            },
            function(error) {
                // functionality error toast
            }
        );
    },

/// some code omit
console.log("This "+name+" id: "+this._searchPaired(name));    // This CSPUSM1411000117 id: undefined

```

is this allowed! Can you please suggest alternative

mumairali, 2016-05-09:
I have tried

```
_searchPaired : function(name) {
        var deviceId = "";
        bluetoothSerial.list(
            function(devices) {
                devices.forEach(function(device) {
                    if (device.name === name) {
                        // found in paired
                        deviceId = device.id;

                    } else {
                        deviceId = "";
                    }   // else
                });
            },
            function(error) {
                // functionality error toast
            }
        );
    },
```
no success

don, 2016-05-09:
Calling bluetoothSerial.list then bluetoothSerial.connect is a good approach.

Can you paste the details of the device here? Add this line before you return device.id

    console.log(JSON.stringify(device));

mumairali, 2016-05-10:
Response is exactly looks like this,

```
[{
	"class": 276,
	"id": "12:CF:48:26:00:01",
	"address": "12:CF:48:26:00:01",
	"name": "M141117"
},{
       ...
}]
```
Can we use Promise to get value to async task `bluetoothSerial.list()` and resolve them anywhere

don, 2016-05-27:
That's good that you're getting results. It means there's an error in the original function you pasted. 

Since you have nested functions, your logic is going to need a promise or callback to return data from the  nested function.

closed 2016-06-23 by don