don/BluetoothSerial #230: Additional Devices iOS

zindello, 2016-05-18:
Hi,

I was wondering if it would be possible to add support for additional devices in iOS, such as the BC118 module as used in the SparkFun BLE Mate 2 module.

They do have support for serial comms with this module via a supplied XCode library, but I'm not up to speed with Objective C and I was hoping to use something like this.

Can you point me in the right direction here?

Cheers,
Josh

don, 2016-05-18:
The best way to support multiple BLE devices is to use my other plugin https://github.com/don/cordova-plugin-ble-central.

See the [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) for one way to do serial connections. Just update the service and characteristic UUIDs to match BLE Service running on the BC118.

zindello, 2016-05-20:
Thanks. I've just moved to one of the devices compatible with this plugin. I'm going to raise another issue I've encountered, it's probably me but some guidance would be appreciated there. I'll close this one off.

closed 2016-05-20 by zindello