don/BluetoothSerial #232: subscribeRawData() partitioned data above than 1008 characters

mumairali, 2016-05-23:
I am sending a command to device which reply me data with more than 3000 characters. I have noticed that command reply is accurate in putty console but when I parse response to json than it become invalid.

here is code
```
                     bluetoothSerial.write(
                        '{"command":"access_points"}',
                        function() {
                            bluetoothSerial.subscribeRawData(
                                function (data) {
                                    console.log("Before Parsing to JSON");
                                   var bytes = new Uint8Array(data);
                                   console.log(String.fromCharCode.apply(null, bytes));     
                                },
                                function () {
                                    console.log("Command Failed")
                                }
                            );
                        });
```

it responsed with this pattern

```
Before Parsing to JSON
_1008 characters string     (Consider this is 1 part of JSON string)_
Before Parsing to JSON
_1008 characters string     (Consider this is 2 part of JSON string)_
Before Parsing to JSON
_remaining part of JSON string_
```

Can you please suggest me what should I do in this scenario!

mumairali, 2016-05-24:
I solved it by introducing a function to check if valid json response other wise add next portion to previous one.

```

var isValidJson = (str) => {
  try {
    JSON.parse(str);
  } catch (e) {
    return false;
  }
  return true;
}

bluetoothSerial.write(
   '{"command":"access_points"}',
   function() {
      var data = "";
      bluetoothSerial.subscribeRawData(
           function (data) {
              data += data; // concat until isValidJson return true
              var str = String.fromCharCode.apply(null, new Uint8Array(data)));
              if (isValidJson(str)) {
                 console.log(str); // <= this is exact what I want :-)
              }
            },
            function () {
              console.log("Command Failed")
            });
 });
```

don, 2016-05-27:
Thanks for the update. Glad you solved this.

closed 2016-05-27 by don