don/BluetoothSerial #233: connection problem with elm327

kauchemar, 2016-05-27:
i'm actually developping a cross-plateform app but i have a problem with elm 327  i use bluetoothSerial.write to send commands such as( 010C,010F) , but when i try to get the recived data with blutoothserial.read() all what i get is " undefined" , so can any one have an idea about this help me i nedd to know how shoud i send commands and howa i can get the response data

don, 2016-05-27:
@kauchemar this sounds like an implementation problem with the ODBII or the ELM327 rather than problem with the plugin.

Try sending the commands in an ArrayBuffer not as Strings or numbers. 

Check the Stackoverflow and other forums for more details about working with ODBII.

closed 2016-05-27 by don

kauchemar, 2016-05-27:
thanks , i'll try sending the caommands in an arraybuffer  .
this is a part of my  old code 
$("#setprotocole").click(function(){
          // send command 
          var num= "ATSP0\r";
          
          bluetoothSerial.write(num, function(){
              alert(num+ "succés d'envoie")}, function(){
              alert(num+"echec d'envoie")});
          
          
         bluetoothSerial.subscribeRawData(function (data) {
 
    alert(data);
}, function (){alert("no data");});
          
       bluetoothSerial.available(function (numBytes) {
        alert("There are " + numBytes + " available to read.");
},  function (){alert("no data");});

          
      });