don/BluetoothSerial #236: subscribeRawData only receives data for a couple of seconds

elafargue, 2016-06-03:
Hi Don!

... I am working with a Bluetooth SPP device which uses a binary protocol, sending large chunks of data (about 3kB) every 5 seconds. Unfortunately, it looks like using subscribeRawData, I only get a couple of bytes, then nothing, though I know the device is sending data continuously... Are there specific steps to follow to read raw data reliably ? Are there known issues with that method?

After subscribeRawData stops triggering its callback, if I call bluetoothSerial.available, I can see that there are 3 bytes available in the buffer. Doing a "read" returns an empty string, and the callback of subscribeRawData is never triggered for those 3 bytes... Any idea what I can do to debug further?

elafargue, 2016-06-03:
OK, I think I solved this: apparently, Android bluetooth sockets don't like to be idle. The device sent data every 5 seconds only, which lead to the link just stopping to receive without an error.

Updating my implementation to ping the BT device every second solved the issue entirely. It looks like as long as you send data on the link at least more often than every 5 second, you're good. Might be my particular device, but I thought this might help others at some point.

venatiodecorus, 2016-06-05:
Not sure if this is related, but I had code that was working fine with subscribeRawData, and last I knew it was working. Then I implemented https://github.com/randdusing/cordova-plugin-bluetoothle for BLE support on iOS. When I got back to testing, I cannot read data on Android anymore. I can scan, and connect to the device fine. But subscribeRawData will not return data or an error. I've been pulling my hair out trying to figure this out. I'm testing on the latest version of Android on a Nexus 6. Latest version of all tools, as I just set them all up on my desktop today.

elafargue, 2016-06-05:
Oh, interesting, I also use that plugin... anyway, try the fix I describe
above, does it fix it for you?

(phone)

venatiodecorus, 2016-06-05:
Yes I tried that, I write a byte to the device every two seconds. Still doesn't seem to work for me.

venatiodecorus, 2016-06-05:
Oh hmm I actually rolled back to an older version and I'm still not getting any data, so there must be some other issue. But still weird that I wouldn't get an error.

venatiodecorus, 2016-06-06:
I figured it out, I had the device set to 115k but it needed to be set to 9600......whoops...darn confusing labels.

elafargue, 2016-06-06:
...is there any way to specify the serial speed from this library?

venatiodecorus, 2016-06-09:
I don't think so. This was a switch on the Bluetooth module, unrelated to the library.

don, 2016-06-23:
@elafargue AFAIK android doesn't let you se the serial speed for SPP

closed 2016-06-23 by don

elafargue, 2016-06-23:
Yes, I investigated - there are APIs missing at the RFCOMM level on
Android for setting the serial speed. In theory it would be possible to
re-implement using the L2CAP APIs but certainly not easy. This means in
particular that the Adafruit EZ-Link cannot work reliably on Android unless
you're lucky and its default serial speed matches what you need - that
module tries to be too smart for its own good and does away with "AT-style"
configuration options...