don/BluetoothSerial #237: How to pair or unpair bluetooth device with device id

korcstar, 2016-06-06:
I want to pair or unpair bluetooth device with device id
I can not find function for this
Please help me

mumairali, 2016-06-22:
@korcstar @noobsoft to **Pair** only you have to call method [`bluetoothSerial.discoverUnpaired()`](https://github.com/don/BluetoothSerial#discoverunpaired) this method will return device IDs then call [`bluetoothSerial.connect()`](https://github.com/don/BluetoothSerial#connect) pass your discoverd _device id_ to `connect()` method simple! Go through example projects

But I have same question how to **unpair** a device? if anyone has idea

don, 2016-06-22:
There's no unfair method. I manage Bluetooth Classic devices via the Android Bluetooth Settings. The discoverUnpaired functionality was donated via a pull request.

This [other bluetooth plugin](https://github.com/tanelih/phonegap-bluetooth-plugin) has a bunch of functions for managing Bluetooth. It has an [unpair function](https://github.com/tanelih/phonegap-bluetooth-plugin/blob/master/www/bluetooth.js#L239). You can probably use this plugin at the same time as my plugin. Alternately, take a look at that code and submit a pull request to add the functionality here.

don, 2016-09-17:
You could also see pull request #114, it has unpair. I haven't merged that pull request because it uses non-public APIs.

closed 2016-09-17 by don