don/BluetoothSerial #240: Print chinese characters

t-nonque, 2016-06-14:
Printing Chinese characters doesn't work. Somebody knows what is the way to make it work?

bluetoothSerial.write("你好", function(){}, function(){});

Prints " '}" on my printer, instead of "你好"

Thanks by advance ^^

don, 2016-06-23:
When you pass a String with unicode characters to `bluetoothSerial.write` it isn't being converted to bytes correctly.

The solution is to convert the string to bytes yourself and pass an array of bytes or an ArrayBuffer to `bluetoothSerial.write`

don, 2016-06-23:
bluetoothSerial.write(stringToBytes("你好"))

Try something like [this function](https://github.com/chariotsolutions/phonegap-nfc/blob/master/www/phonegap-nfc.js#L567-L596) to convert

    var stringToBytes = function(string) {
        // based on http://ciaranj.blogspot.fr/2007/11/utf8-characters-encoding-in-javascript.html

        var bytes = [];

        for (var n = 0; n < string.length; n++) {

            var c = string.charCodeAt(n);

            if (c < 128) {

                bytes[bytes.length]= c;

            } else if((c > 127) && (c < 2048)) {

                bytes[bytes.length] = (c >> 6) | 192;
                bytes[bytes.length] = (c & 63) | 128;

            } else {

                bytes[bytes.length] = (c >> 12) | 224;
                bytes[bytes.length] = ((c >> 6) & 63) | 128;
                bytes[bytes.length] = (c & 63) | 128;

            }

        }

        return bytes;
    };

closed 2016-08-09 by don

skenuo, 2018-03-04:
hi!.
Print Chinese, or garbled.

Gunasekaran15, 2018-04-07:
I tested it but it does't print chinese and other char it prints symbols

don, 2018-05-02:
@Gunasekaran15 it's possible the data needs to be encoded some other way for the printer. Maybe UTF-16? Getting the text into an ArrayBuffer will stop the plugin from messing with your data. The trick is figuring out what format the printer is expecting. Hopefully your printer manufacturer can help out with the details. If you get it working please post the result here to help other people.

/cc @rakesanug

zpaaa, 2018-06-29:
utf8->gbk first

Ng-India99, 2019-06-04:
Hi,

i followed your steps.

I can able to print the text, Arabic characters but i am unable to print the formatted text like align center, bold text etc

how to pass ESC/POC specific commands to plugin ?

i tried like below for bold text but its not working.

TXT_BOLD_ON: '\x1b\x45\x01';

// BoldOn hex code=0x2fa00bf0e86c440658a6a71;

 const data = new Uint8Array([0x2fa00bf0e86c440658a6a71]);
   this.printer.printData(data);

[image]

B200019A, 2023-04-19:
hi, i also face cannot print the chinese text in printer, everyone can give me suggestion
 var text = "你好\n";
   var encodeText = new TextEncoder("GB18030", {
    NONSTANDARD_allowLegacyEncoding: true,
  }).encode(text + "\r\n\u001B\u0070\u0000\u000F\r\n");

  ble.write(
    printer_id,
    "49535343-FE7D-4AE5-8FA9-9FAFD205E455",
    "49535343-8841-43F4-A8D4-ECBE34729BB3",
    text.buffer,
    (success) => {
      console.log(success + ": print" + text);
    },
    (failure) => {
      console.log(failure);
    }
  );>