don/BluetoothSerial #243: Raw data not working correctly

acondurache, 2016-06-30:
I am using a Force Sensor and trying to access the data. In the Arduino Serial Monitor I get the following data:

```
Resistance: 3301060.00 ohms
Force: 0.47 g

Resistance: 3301060.00 ohms
Force: 0.47 g

```
...

But when I access the data via  **bluetoothSerial.subscribeRawData()** using code below:
```
 bluetoothSerial.subscribeRawData(function (data) {
            var bytes = new Uint8Array(data);
            console.log(bytes);
        }, $scope.failConnect);
```

I get a bunch of random arrays that doesn't correspond to the data I get above from the Arduino Serial Monitor

Any ideas why I am not getting the correct data?

closed 2016-06-30 by acondurache

don, 2016-06-30:
@acondurache did you solve this? what was the problem?

Purushothama-VS, 2017-10-11:
Hi Don,
I am also facing similar issue and getting similar data like above, how to resolve this.