don/BluetoothSerial #244: Adding New buttons doesn't work

LingboTang, 2016-07-07:
I'm new to this plugin and I'm playing with the SimpleSerial Example.
Basically I just replaced the "Connect" button to a "Scan" button to discover the unpaired devices and return the list of them.

I added a "scanButton" in index.html

```
<!DOCTYPE html>

<html>
    <head>
        <title>Simple Serial</title>
    </head>

    <body>
        <div class="app">
            <div id="ui">
                <!--Scan Here-->
                <button id="scanButton">Scan</button>
            </div>
        <div id="message">Hello.</div>
        </div>
        <script type="text/javascript" src="cordova.js"></script>
        <script type="text/javascript" src="js/reader.js"></script>
        <script>
            app.initialize();
        </script>
    </body>
</html>
```

Then replaced the control in reader.js:
```
var app = {
    macAddress: "00:0B:CE:07:36:AB",  // get your mac address from bluetoothSerial.list
    chars: "",

    initialize: function() {
        this.bindEvents();
        console.log("Starting SimpleSerial app");
    },

    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
        scanButton.addEventListener('touchend', app.manageScan, false);
    },

    onDeviceReady: function() {
        // check to see if Bluetooth is turned on.
        // this function is called only
        //if isEnabled(), below, returns success:
        alert('Before calling pairing...');

        var unPaired = function() {
            bluetoothSerial.discoverUnpaired(
                function(results) {
                    app.display(JSON.stringify(results));
                },
                function(error) {
                    app.display(JSON.stringify(error));
                }
            );
        }

        // if isEnabled returns failure, this function is called:
        var notEnabled = function() {
            app.display("Bluetooth is not enabled.")
        }

         // check if Bluetooth is on:
        bluetoothSerial.isEnabled(
            unPaired,
            notEnabled
        );
    },

    manageScan: function() {
        alert("events Listening");
        app.clear();
        app.display("Scanning available Devices...");

        bluetoothSerial.discoverUnpaired(
            function(results) {
                app.display(JSON.stringify(results));
            },
            function(error) {
                app.display(JSON.stringify(error));
            }
        );
    }

    
    showError: function(error) {
        app.display(error);
    },

    display: function(message) {
        var display = document.getElementById("message"), // the message div
            lineBreak = document.createElement("p"),     // a line break
            label = document.createTextNode(message);     // create the label

        display.appendChild(lineBreak);          // add a line break
        display.appendChild(label);              // add the message node
    },

    clear: function() {
        var display = document.getElementById("message");
        display.innerHTML = "";
    }
};      // end of app

```
But I couldn't even get the the alert when devices is ready, and as well I couldn't get the alert in manageScan, which means the event didn't get listened. Does anybody know what I'm missing?

don, 2016-07-07:
There's an error in your JavaScript. You're missing a comma at the end of manageScan. I usually find errors like this by running jshint.

    $ jshint www/js/reader.js
    www/js/reader.js: line 30, col 10, Missing semicolon.
    www/js/reader.js: line 34, col 53, Missing semicolon.
    www/js/reader.js: line 35, col 10, Missing semicolon.
    www/js/reader.js: line 60, col 5, Expected '}' to match '{' from line 1 and instead saw 'showError'.
    www/js/reader.js: line 60, col 14, Missing semicolon.
    www/js/reader.js: line 60, col 5, Unrecoverable syntax error. (77% scanned).
    
    6 errors
    $ 

You can also see the error using chrome://inspect

[image]

closed 2016-07-07 by don

LingboTang, 2016-07-07:
Yes, my bad, I'm debugging with logcat, it's a little harder to read the long log.