don/BluetoothSerial #245: Async Issue of subscribeRawData

LingboTang, 2016-07-12:
I got some problems implementing subscribeRawData:

And I'm guessing subscribeRawData is not asynchronous.

```
bluetoothSerial.write('M,0\rWB\rVV\rM,433\r'
            , function(data){
            console.log('in write success function:');

            bluetoothSerial.subscribeRawData(function (data) {
                console.log('in subscribeRawData success');
                var bytes = new Uint8Array(data);
                console.log('in coverting to bytes');
                console.log(bytes);
                var str= String.fromCharCode.apply(null, bytes);
                app.display(str);
                console.log(str);
            }, function(error){
                console.log('in subscribeRawData failure');
            });
            
        }, function(error){
            console.log('in write error function');
            console.log(error);
        });
```
I'm giving an example command in my write function and try to subscribe the raw response from the RF reader. If success, it should return VV,'firmware-version','buildid','modelid' as one line. But what I get from the str is fragiled and randomly tokenized.

don, 2016-07-13:
One problem is your sample code keeps subscribing for raw data every time you write data. You should only call subscribeRawData _one_ time. The success callback, `onData` in this case, will be called _every_ time data is received. Writing and receiving data are completely disconnected.

    // this callback is called every time data is received
    function onData = function (data) {
        console.log('received data from peripheral');
        var bytes = new Uint8Array(data);
        console.log('in coverting to bytes');
        console.log(bytes);
        var str= String.fromCharCode.apply(null, bytes);
        app.display(str);
        console.log(str);
    }

    // only subscribe 1 time
    bluetoothSerial.subscribeRawData(
        onData, 
        function(error){
            console.log('in subscribeRawData failure');
        }
    );

    bluetoothSerial.write(
        'M,0\rWB\rVV\rM,433\r', 
        function(data){
            console.log('sent data to peripheral');
        }, function(error){
            console.log('error sending data ');
            console.log(error);
        }
    );

LingboTang, 2016-07-13:
It is my reader that cut off the message, I am concatenating them after I received the entire response. But your approach is a neat shoot, thank your for pointing it out. I think I can close this issue now.

closed 2016-07-13 by LingboTang