don/BluetoothSerial #246: Read/Write after connecting.

aserrin55, 2016-07-15:
I have scanned devices and everything is OK.
But when I try to connect with any device there is no pin requirement like:

[image]

I'm sure that the device is password protected, but the input dialog does not appear.
So, I make the connection and the status is connected, but when I want to read something, there are some characteristics encrypted and its value is not readable. But the "free" characteristics can be read properly. For example:

[image] 

What should i do? Where is the password need to be included? My connection code:
`$scope.connect = function(addressParam) {
      var params = {address: addressParam};

      new Promise(function (resolve, reject) {
        bluetoothle.connect(resolve, reject, params);
      }).then(connectSuccess, handleErrorC);

      function connectSuccess(result) {
        console.log("connectWithExit");
        console.log(result);
      }

      function handleErrorC(result) {
        console.log("ERRORalConectarse");
        console.log(result);
      }
    }`

Thanks

BeyondIntelligence, 2016-07-30:
I have the same issue but with arduino and android.

don, 2016-07-30:
@aserrin55 are you seeing this on iOS or Android? The screenshots look like Bluetooth Low Energy. You're probably better off using https://github.com/don/cordova-plugin-ble-central. With BLE, you should get prompted to bond with a pin (which looks like a Bluetooth classic pairing dialog) when you read a characteristic that has PERMISSION_READ_ENCRYPTED.

don, 2016-07-30:
@BeyondIntelligence with Bluetooth Classic on Android I usually pair my phone with the Bluetooth device using the Android control panel, then call bluetoothSerial.list() to see the device. There is [bluetoothSerial.discoverUnpaired](https://github.com/don/BluetoothSerial#discoverunpaired) to list unpaired devices. Calling connect on an unpaired device should begin the pairing process, which will prompt you for the pin.

closed 2016-09-17 by don