don/BluetoothSerial #248: subscribe old data after reconnection

LingboTang, 2016-07-25:
I've met a problem of device receiving old data:

For example I defined this function here. And every time I disconnect the reader and then reconnect it again, I would receive the data a couple of seconds ago. I'm not sure if the reader was buffering the data or the `bluetoothSerial.subscribe` was caching the data. 
```
getTagLocation: function(tagId,timeout,callback) {
        console.log('in getTagLocation func')
        reader.tagLocation_callback = callback;
        reader.writeCommand(reader.location_command);
        // Subscribe Raw Data
        bluetoothSerial.subscribe(
            '\r',
            function(data) {
                reader.onListeningData(data,tagId);
            },
            function(error){
                console.log(error);
        });
    },
```

don, 2016-08-09:
It's possible the data was cached by subscribe. Can you try setting a break point in native code to determine this?

LingboTang, 2016-08-09:
Yes I can. Let me test this.

LingboTang, 2016-08-26:
@don I think it's my reader was storing the cache. When I send the reset command to my reader, it cleared the buffers, I think it's good to close this issue now.

closed 2016-08-26 by LingboTang