don/BluetoothSerial #249: onData: function(data) Not getting called

BeyondIntelligence, 2016-07-30:
Hi,

I am successfully connecting the Ardunio Uno to my android Nexus 6P device. But I am not able to receive and data. The function   onData: function(data) is never being called. 

I have tried with the GChat application and the data is received from Uno. 

Any inputs are appreciated. Thank you in advance.

don, 2016-07-30:
What Bluetooth radio are you using with the Arduino?

Check adb logcat and chrome://inspect for error messages. 

Can you link to the code you're running on the Uno and the phone?

BeyondIntelligence, 2016-07-30:
Thanks for the prompt reply. 

I am using a Sprakfun BlueSmirf.

There is no error. It just does not call the onData: Function(data). .

Phone : Is the Chat example.
Uno: The example given.

I am pairing the device before hand.

don, 2016-09-17:
I think you need might need to send some more configuration to your device. I have a Sparkfun Bluetooth Mate Silver and I have some additional configuration to get my modem working.

    void setupBluetooth() {
      bluetooth.begin(115200);  // The Bluetooth Mate defaults to 115200bps
      bluetooth.print("$$$");  // Enter command mode
      delay(100);  // Short delay, wait for the Mate to send back CMD
      bluetooth.println("U,9600,N");  // Temporarily Change the baudrate to 9600, no parity
      // 115200 can be too fast at times for NewSoftSerial to relay the data reliably
      bluetooth.begin(9600);  // Start bluetooth serial at 9600
    }

See https://github.com/don/BluetoothSerial/blob/master/examples/Chat/Arduino/Chat_Sparkfun/Chat_Sparkfun.ino#L58-L65

closed 2017-03-17 by don