don/BluetoothSerial #253: Get connected device info.

rjurado01, 2016-08-10:
It is possible to get the connected device info (address, id or name) ?
I don't see any method to get this info.

Thanks.

LingboTang, 2016-08-26:
@rjurado01 You can do it with `bluetoothSerial.list` to list all settings for all devices, or you can also use `bluetoothSerial.showBluetoothSettings`

rjurado01, 2016-09-05:
@LingboTang `bluetoothSerial.list` doesn't return which device is connected and `bluetoothSerial.showBluetoothSettings` doesn't work for me because I need connected device info from code to show: "Device connected: XV9040" in my application.

don, 2016-09-17:
There's no direct way to get the connected device info. You can use @LingboTang's suggestion of using bluetoothSerial.list. Save the MAC address (or UUID) that you pass to bluetoothSerial.connect and filter the results from bluetoothSerial.list to get the name.

closed 2016-09-17 by don