don/BluetoothSerial #255: How bluetoothserial write and read works?

KennyJack, 2016-08-24:
Hi,

I able to connect both android device1 to android device2.
In android device1 i try bluetoothserial.write("hello world").
How can my android device2 receive this "hello world" message?
Please help.

LingboTang, 2016-08-25:
They do have the function:

```
bluetoothSerial.subscribe('\n', function (data) {
            app.clear();
            app.display(data);
        });
```
Subscribe the data use this function please.

LingboTang, 2016-08-26:
@KennyJack Does it solve your problem?

biy-gxnn, 2016-09-01:
do you solve your problem? I've the same problem,but I used subscribe and it didn't work

KennyJack, 2016-09-01:
Hi all,

I haven't solve my problem, i doubt that is device problems as some phone doest not support COM/ serial port. I'm going to bought arduino or omega2 to try out.
if you try out on two android devices and works please tell me. thanks

biy-gxnn, 2016-09-01:
I use on two android devices,samsung and huawei. If it works, I would tell you

mumairali, 2016-09-01:
@biy-gxnn I am curious. Does this work for you android to android?

This is mentioned in plugin _README.md in Limitation section_: **Will not connect Android to Android***

if it will not connect android to android then how you will read and write?

KennyJack, 2016-09-01:
Hi @mumairali ,

But the reason i able to connect both android to android. 
@LingboTang  may be can explain why readme state Limitation section: Will not connect Android to Android*

LingboTang, 2016-09-01:
@KennyJack I haven't tried to connect Android to Android. Metadata traffic between android to car, reader and drones works for me. If that's written in README, I would say there's no need to waste your time trying it out. But they do have a more general function called `bluetoothSerial.subscribeRawData()` which let you listen to the binary data.

```
// the success callback is called whenever data is received
bluetoothSerial.subscribeRawData(function (data) {
    var bytes = new Uint8Array(data);
    console.log(bytes);
}, failure);
```

Make sure you decode them to string and set break points on those received strings when you meet some delimiter such as `\r` carriage return.

LingboTang, 2016-09-01:
@mumairali @biy-gxnn I think @mumairali is right, and for more detailed information please go to this thread: https://github.com/don/BluetoothSerial/issues/50. It's an enhancement and hasn't worked completely yet.

don, 2016-09-17:
See https://github.com/don/BluetoothSerial/issues/260#issuecomment-247744419

closed 2016-09-17 by don