don/BluetoothSerial #259: Ionic Android pairing with Bluetooth without PIN

santoshshinde2012, 2016-08-30:
I am working with bluetooth module. I have a Ionic app that uses the Bluetooth Serial plugin and I can connect to the bluetooth module successfully, however I want to be able to remove the pop up that appears asking the user to enter the PIN to pair my Android to the Bluetooth module.

So its feasible to bluetooth pairing without appears asking the user to enter the PIN .

If yes then told me the way.

Thank you.

don, 2016-09-17:
You might be able to use the Android Bluetooth settings to pair the device before running the app.

If that doesn't work there's an Android API to set the PIN [BluetoothDevice.setPin](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#setPin(byte[])) you might be able to modify the plugin to accept this programatically.

don, 2016-09-17:
It looks like there's pairing with a PIN in pull request #114 

That request uses non-public APIs so I haven't merged it.

mumairofficial, 2017-01-23:
@don It will be great help from you if you provide merged code #114 gist or separate branch. I am looking forward for great news from your side.

don, 2017-02-02:
@mumairofficial I didn't merge that plugin because it uses reflection. Unfortunately since it's old, there are now conflicts. 

If you want this functionality you can use the branch for the pull request from the authors github https://github.com/stakutis/BluetoothSerial/tree/stakutis

Backup your project
Remove the official version of the plugin

     cordova plugin rm cordova-plugin-bluetoothserial

Try the alternate version with the feature you want 

     cordova plugin add https://github.com/stakutis/BluetoothSerial/tree/stakutis

mumairofficial, 2017-02-03:
@don thank you so much. I will try and let you know asap.

closed 2017-03-17 by don

hirenvolansys, 2017-03-23:
@don I want to bond and unbond BLE device. How it is possible in Ionic? I also require PIN popup while connecting, which not come when connect with BLE device. 

Please help me. Thanks in advance.

don, 2017-03-24:
@hirenvolansys it's best to create new issues rather than comment on closed issues.
 
iOS should bond automatically if you try and connect to a device that requires it or read/write a characteristic with enhanced permissions.

There is an old Android pull request for that provided bind and unbind but uses reflection so it won't be merged into master.  See #112 and #114