don/BluetoothSerial #261: RangeError: Maximum call stack size exceeded

mspace, 2016-09-13:
I'm using next code to connect to my bluetooth printer (http://www.rongtatech.com/products-look-27.html) and print invoice/string: 
Using:

- Windows 10 x64
- Visual Studio 2015
- Samsung galaxy xcover 3 running Android 4.4.4 kitkat (http://www.samsung.com/uk/consumer/mobile-devices/smartphones/others/SM-G388FDSABTU)
- package.json
```javascript
{
  "name": "TestCordovaKendo",
  "version": "1.0.0",
  "private": true,
  "dependencies": {
    "gulp": "3.9.0",
    "gulp-print": "1.1.0",
    "bower": "1.4.1",
    "gulp-less": "3.0.3",
    "gulp-rename": "1.2.2",
    "cordova-lib": "5.4.1",
    "gulp-typescript": "2.10.0",
    "cordova": "5.4.1",
    "gulp-cordova-version": "0.3.0"
  }
}
```

Often I get maximum call stack size exceeded, stoped at **cordova.js:**. Using **debug mode, runnign with disabled all breakpoints** Have I missed something important? 

```javascript
    /**
     * Called by native code when returning the result from an action.
     */
    callbackFromNative: function(callbackId, isSuccess, status, args, keepCallback) {
        try {
            var callback = cordova.callbacks[callbackId];
            if (callback) {
                if (isSuccess && status == cordova.callbackStatus.OK) {
                    callback.success && callback.success.apply(null, args);
                } else if (!isSuccess) {
                    callback.fail && callback.fail.apply(null, args);
                }
                /*
                else
                    Note, this case is intentionally not caught.
                    this can happen if isSuccess is true, but callbackStatus is NO_RESULT
                    which is used to remove a callback from the list without calling the callbacks
                    typically keepCallback is false in this case
                */
                // Clear callback if not expecting any more results
                if (!keepCallback) {
                    delete cordova.callbacks[callbackId];
                }
            }
        }
        catch (err) {
            var msg = "Error in " + (isSuccess ? "Success" : "Error") + " callbackId: " + callbackId + " : " + err;
            console && console.log && console.log(msg);
            cordova.fireWindowEvent("cordovacallbackerror", { 'message': msg });
            throw err;
        }
    },
```
I'm attaching my android_debug_log which
[visual_studio_android_debug.zip](https://github.com/don/BluetoothSerial/files/469757/visual_studio_android_debug.zip)

My service for printing:
```javascript

var Services2 = {
  PrintService2: {
[... 253 lines cut ...]
    stringToAsciiArray: function (data) {

      var ar = [];
      for (var i = 0; i < data.length; i++)
        ar.push(data.charCodeAt(i));

      return ar;
    },
    rightPadding: function (str, length) {

      while (str.length < length)
        str = str + " ";

      return str;
    },
    leftPadding: function (str, length) {

      while (str.length < length)
        str = " " + str;

      return str;
    }
  }
}
```

don, 2016-09-20:
@mspace You hit the call stack limit. Usually this is bad logic in some recursive calls, but it can be caused by UI Binding problems, or importing the same js file 2x.

Can you use chrome://inspect and view the stack when it blows up?

I looked at your code and log file and didn't see anything obvious.

mspace, 2016-09-26:
Thank you for your feedback. Didn't know about chrome://inspect options, will look into that when debugging.

mspace, 2016-09-29:
I noticed that the code works (printer successfully prints everything as it should) when I set a breakpoint somewhere in the code in visual studio. Right after I delete or disable all breakpoints printing stops. Does this have to do something with device receiveing data to fast? I can't explain such behaviour. Interesting is that using this code, it works ok (even if I delete all breakpoints)? Maybe something wrong with javascript promises? If I nest functions it works without promises. Do you have any idea?

```javascript
      // 1. initialize printer: Clears the data in the print buffer and resets the printer mode to the mode that was in effect when the power was turned on
      bluetoothSerial.write([0x1b, 0x40], function (success1) {

        // 2. set normal font size 12x12
        bluetoothSerial.write([0x1b, 0x21, 0x00], function (success2) {

          // 3. set character code table to ISO-8859-2[Latin 2]: [27, 116, 36]
          bluetoothSerial.write([0x1b, 0x74, 0x24], function (success3) {
```

closed 2017-03-17 by don