don/BluetoothSerial #263: Initiate interrupt / timeout if SPP device failed to respond after specific time interval.

mumairali, 2016-09-19:
I am working on bluetooth communication with SPP (Serial Port Profile) based device and There is scenario:

I have sent some commands to my device `write()` and `SubscribeRawData()` and waiting for response. In case device failed to respond (firmware crashed or hanged) then my app will stuck in infinite loop of waiting in `SubscribeRawData()` function.

Is there any way to tackle this situation of infinite loop?

mumairali, 2016-11-23:
@don I am using JavaScript Promise pattern. How can I handle this situation if data is not arrived?

don, 2017-01-05:
See #279 for an explanation of why you can't use Promises here.

You should add the subscriber one time before writing any data and leave the subscriber connected.

closed 2017-01-05 by don