don/BluetoothSerial #264: Android Build fails with the latest Version

vrajesh126, 2016-09-20:
Hey Don,

I installed current version of this plugin onto my project with telerik AppBuilder and it failed to build with the following message

Error: exit code: 1
    at Error (native)
    at ChildProcess.<anonymous> (/home/builder/BpcTooling/Cordova3/node_modules/cordova-build-utils/bin/install-plugins-3.js:190:14)
    at emitTwo (events.js:87:13)
    at ChildProcess.emit (events.js:172:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
	Error: Build failed during execution
	Error: 'Build failed with error code 1'
Done building project "LSIDeveloper.tmp.proj" -- FAILED.
.

I checked the Id  "com.megster.cordova.bluetoothserial" and it is supposed to be that.  I am using Cordova Version 4.0.0 for App Builder.

Appreciate your help on this.

don, 2016-09-21:
You're using the old plugin id, so that code is probably incompatible with the latest Android. Try replacing it with cordova-plugin-bluetooth-serial.

    cordova plugin rm com.megster.cordova.bluetoothserial
    cordova plugin add cordova-plugin-bluetooth-serial

vrajesh126, 2016-09-21:
Thank you Don! I did change the ID to the current as suggested, but stills give me the same error message. And it also seems like some other user has the same issue. 

Appreciate your help on this.

don, 2016-09-21:
@vrajesh126 what error message are you getting now?

Can you try creating a temp project and see if you get the same error?

Here's the steps I used to create a new project and it compiled without errors. 

    $ cordova -v
    6.3.1

    $ cd /tmp
    $ cordova create foo
    Creating a new cordova project.
    $ cd foo
    $ cordova platform add android
    Adding android project...          
    $ cordova plugin add cordova-plugin-bluetooth-serial
    Fetching plugin "cordova-plugin-bluetooth-serial" via npm
    Installing "cordova-plugin-bluetooth-serial" for android
    $ cordova run android --device

vrajesh126, 2016-09-21:
Will Cordova Version affect the Plugin? Do i need latest version for Cordova ? As Telerik app builder  would allow me to update the cordova version up to 5.2.0. and not to 6.3.1

don, 2016-09-21:
@vrajesh126 it looks like the latest won't work with Cordova 5.2.0 due to the new android-23 permissions check.

Can you use version 0.4.5 of the plugin? That will be compatible with Cordova 5.

gerarrd, 2016-09-22:
Hello there, I am trying to incorporate bluetooth serial to the application I am building and I seem to encounter the same problem, below is the last lines of the error log

`:modules:AGCordova:compileReleaseJava/Users/administrator/app-builder/STAGING/client/modules/AGCordova/src/com/megster/cordova/BluetoothSerial.java:216: error: cannot find symbol
            if (cordova.hasPermission(ACCESS_COARSE_LOCATION)) {
                       ^
  symbol:   method hasPermission(String)
  location: variable cordova of type CordovaInterface
/Users/administrator/app-builder/STAGING/client/modules/AGCordova/src/com/megster/cordova/BluetoothSerial.java:220: error: cannot find symbol
                cordova.requestPermission(this, CHECK_PERMISSIONS_REQ_CODE, ACCESS_COARSE_LOCATION);
                       ^
  symbol:   method requestPermission(BluetoothSerial,int,String)
  location: variable cordova of type CordovaInterface
/Users/administrator/app-builder/STAGING/client/modules/AGCordova/src/com/megster/cordova/BluetoothSerial.java:467: error: method does not override or implement a method from a supertype
    @Override
    ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
3 errors
 FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':modules:AGCordova:compileReleaseJava'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1 mins 0.025 secs`

are you perhaps familiar with appgyver's supersonic? I am using that framework to build the application.

Thank you!

bitbandit01, 2016-09-22:
Hey gerarrd,
I was getting exactly the same message as you while trying to integrate this into my Ionic 1.3.1 project.
However, I seem to have been able to solve it by running the following:
`cordova platform update android`

I think that cordova-android@5.0.0 or above is required to correctly implement the new Android permissions stuff.  Hope this helps!

JosephDAM, 2016-09-22:
If you are using an old version of cordova you should try `cordova plugin add https://github.com/don/BluetoothSerial#0.4.5 --save` which solved my issue.

Regards

gerarrd, 2016-09-23:
yes, I think the issue lies with the current version of bluetooth serial being not compatible with the version of cordova used by the framework. I opted to use the ver 0.4.5 as you've suggested and it works fine now.

Thank you!

Vraj126, 2016-09-28:
Thanks, Solution did work, i downgraded to 0.4.5 and my project builds up.

vrajesh126, 2016-09-28:
Thanks, Solution did work, i downgraded to 0.4.5 and my project builds up.

closed 2016-09-28 by vrajesh126