don/BluetoothSerial #265: Android build error

JosephDAM, 2016-09-21:
Hi don!

Good ol' copy paste error dump:

`Compiling 5 source files to C:\inetpub\wwwroot\360forMobile\platforms\android\ant-build\classes
    [javac] warning: [options] source value 1.5 is obsolete and will be removed in a future release
    [javac] warning: [options] target value 1.5 is obsolete and will be removed in a future release
    [javac] warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.
    [javac] C:\inetpub\wwwroot\360forMobile\platforms\android\src\com\megster\cordova\BluetoothSerial.java:216: error: cannot find symbol
    [javac]             if (cordova.hasPermission(ACCESS_COARSE_LOCATION)) {
    [javac]                        ^
    [javac]   symbol:   method hasPermission(String)
    [javac]   location: variable cordova of type CordovaInterface
    [javac] C:\inetpub\wwwroot\360forMobile\platforms\android\src\com\megster\cordova\BluetoothSerial.java:220: error: cannot find symbol
    [javac]                 cordova.requestPermission(this, CHECK_PERMISSIONS_REQ_CODE, ACCESS_COARSE_LOCATION);
    [javac]                        ^
    [javac]   symbol:   method requestPermission(BluetoothSerial,int,String)
    [javac]   location: variable cordova of type CordovaInterface
    [javac] C:\inetpub\wwwroot\360forMobile\platforms\android\src\com\megster\cordova\BluetoothSerial.java:467: error: method does not override or implement a method from a supertype
    [javac]     @Override
    [javac]     ^
    [javac] 3 errors
    [javac] 3 warnings`

I have not built for like one or two weeks, and suddenly I get this error :/
If I uninstall your plugin the app is built, but I really like your plugin!!

Regards

don, 2016-09-21:
What version of Cordova are you using?

The latest build adds support for Android-23 permissions so you'll need a recent version of Cordova. I'm using Cordova 6.3.1. For older project you might need to upgrade or regenerate the Android platform code.

don, 2016-09-21:
This might be a duplicate of #264

JosephDAM, 2016-09-22:
`cordova plugin add https://github.com/don/BluetoothSerial#0.4.5 --save`

Works like a charm :+1: 

Thank you Don :)

closed 2016-09-22 by JosephDAM