don/BluetoothSerial #268: setDeviceDiscoveredListener a lot of duplicate devices found

mspace, 2016-09-23:
Using android 5.1.1 with next javascript code:

```javascript
    onShow: function (e) {

      var addDeviceToList = function (d) {
        
        if( viewModel.get("devices").length < 0 ) {
          viewModel.set("devices", [d]);
          return;
        }

        for (var i = 0; i < viewModel.get("devices").length; i++) {
          if (viewModel.get("devices")[i].address == d.address) {
            return; // device already in array, skip
          }
        }

        // not in devices array, add it
        var devicesArr = viewModel.get("devices");
        devicesArr.push(d);
        viewModel.set("devices", devicesArr);
      }

      // 1. list bluetooth devices, if isEnabled returns failure, this function is called:
      var notEnabled = function () {
        alert("Bluetooth is disabled");
      }

      var listBondedDevices = function () {

        viewModel.set("progress", "Searching bonded devices ...");

        bluetoothSerial.list(function (devices) {

          var d = [];
          devices.forEach(function (device) {

            // add to list
            d.push({
              "name": device.name,
              "address": device.address,
              "id": device.address,
              "class": device.class
            });

            console.log(device.id);
          });

          if (d.length > 0)
            viewModel.set("devices", d);

          discoverUnpairedDevices();

        }, showError);
      }

      var discoverUnpairedDevices = function () {

        viewModel.set("progress", "Searching unpaired devices ...");

        bluetoothSerial.setDeviceDiscoveredListener(function (device) {
          // the same scanned device is found multiple times
          console.log('Found: ' + device.id);
          addDeviceToList(device);
        });

        bluetoothSerial.discoverUnpaired(function (unpairedDevices) {

          bluetoothSerial.clearDeviceDiscoveredListener();

          viewModel.set("progress", "Found devuces: " + viewModel.get("devices").length);
        }, failure);
      }

      var showError = function () {
        alert("Error while listing bonded devices");
      }

      var failure = function () {
        alert("Error while discovering unpaired bluetooth devices");
      }

      // clear list
      viewModel.set("devices", []);

      bluetoothSerial.isEnabled(listBondedDevices, notEnabled);
    },
```

Console log:

> > Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 00:02:0A:04:4D:A7
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4
Found: 40:2C:F4:66:4D:E4

don, 2016-09-27:
Those might be Bluetooth Low Energy devices showing up in the scan. For now you'll just need to ignore them, or perhaps store the scan results in a Map keyed by MAC address before displaying to the user.

mspace, 2016-09-27:
I'am already checking duplicated MAC addresses:

```javascript
 for (var i = 0; i < viewModel.get("devices").length; i++) {
   if (viewModel.get("devices")[i].address == d.address) {
     return; // device already in array, skip
  }
}
```

Ok, thank you for now.

closed 2016-09-27 by mspace