don/BluetoothSerial #272: Android to android connection

htahir6200, 2016-10-22:
Hello everyone. A very good material for serial connection. I am using Samsung note 2 and Samsung J1 ace, but i am not able to create a serial connection between them, I have followed the same steps mentioned by the author but after installing app , i am getting this error on both devices.
ERROR: Unable to connect to device

How can i resolve this problem, anyone, please share your comments
Looking towards your response.

Sundypha, 2016-11-03:
Please see https://github.com/don/BluetoothSerial#limitations.

Android to Android is not supported.

don, 2016-11-04:
@htahir6200 There are people who have forked the plugin and added the ability to listen for RFCOMM connections on Android.  I'm hoping that someone solves this and sends a pull request so it can be included in the main plugin. I would be a great feature to have, but I haven't needed it for a project, so I haven't implemented it.

closed 2016-11-04 by don