don/BluetoothSerial #275: Can read but can't subcribe

funreadsapp, 2016-11-18:
Hello,

i am having an issue subscribing. i have a raspberry pi and android connected together. i can connect and read data but when i try to subscribe the function call doesnt call the success or error methods i defined. any idea why that would happen?

thank you

don, 2016-11-18:
What does your subscribe code look like?

Subscribe will only call the success callback when it encounters the specified token in the data. Looking at the example for [subscribe](https://github.com/don/BluetoothSerial#subscribe), if the data doesn't contain a new line \n, the success callback will never be called. Replace the delimiter with a token than you're expecting in your data stream.

    bluetoothSerial.subscribe('\n', function (data) {
        console.log(data);
    }, failure);

funreadsapp, 2016-11-18:
you're absolutely right, i wasn't passing in my delimiter. thank you so much Don!

closed 2016-11-18 by funreadsapp

storm1er, 2017-01-31:
Sorry about this, but @funreadsapp how did you make this to work on raspberry pi ?

funreadsapp, 2017-01-31:
I have python code running on the pi from this [tutorial](https://people.csail.mit.edu/albert/bluez-intro/x232.html) 
example 3-2 is the one you would use to make the pi wait for a connection. you will still need to do some configurations to the pi (like making it discoverable and maybe a few other things depending on what you're making) 
once i have the "client_sock" connected will send what i need like this
`client_sock.send(data + "\n")`

then on the android side i subscribe to "\n" and then do what i need