don/BluetoothSerial #276: Scrambled serial data returned using subscribe() on iOS

dpa99c, 2016-11-18:
I’m using this plugin to communicate with a [Satlab SLC GNSS receiver](http://www.satlabgps.com/en/products/gis/slc.aspx) which contains a [Laird BT-900](http://www.lairdtech.com/products/bt900-series) chip.

The SLC returns a stream of NMEA data over the Bluetooth connection, the delimiting character being "\n" between NMEA sentences, hence I'm doing:

    bluetoothSerial.subscribe('\n', onSuccess, onError);

On Android, it works fine with one NMEA string returned on each invocation of `onSuccess()`, e.g.

    GPZDA,172813.00,18,11,2016,,*64
    $GPGSV,3,1,12,30,72,115,52,05,71,265,51,07,47,060,44,13,38,270,38*7F
    $GPGSV,3,2,12,20,21,308,,02,19,217,,28,17,140,45,09,17,088,32*79
    $GPGSV,3,3,12,15,08,274,,21,05,335,,49,32,171,38,33,31,197,*79
    $GPRMC,172813.00,A,5044.9905982,N,00148.1128307,W,0.055,199.0,181116,0.0,E,D*23
    $GPGSA,M,3,30,05,28,07,09,13,,,,,,,2.5,1.6,1.9*36
    $GPGGA,172813.00,5044.9905982,N,00148.1128307,W,9,06,1.6,7.217,M,49.30,M,07,0136*47
    $GPGLL,5044.9905982,N,00148.1128307,W,172813.00,A,D*70
    $GPVTG,198.998,T,198.998,M,0.055,N,0.102,K,D*25
    $GPGST,172813.00,1.32,1.89,0.99,-4.5539,1.88,1.00,2.47*4B
    $GPZDA,172814.00,18,11,2016,,*63
    $GPGSV,3,1,12,30,72,115,52,05,71,265,51,07,47,060,44,13,38,270,39*7E
    $GPGSV,3,2,12,20,21,308,,02,18,217,,28,17,140,45,09,17,088,32*78
    $GPGSV,3,3,12,15,08,274,,21,05,335,,49,32,171,39,33,31,197,*78
    $GPRMC,172814.00,A,5044.9906038,N,00148.1128502,W,0.029,14.4,181116,0.0,E,D*17
    $GPGSA,M,3,30,05,28,07,09,13,,,,,,,2.5,1.6,1.9*36
    
    
But on iOS, I'm getting data which contains overlapping NMEA sentences, such as:

    $GPGSV,4,2,13,09,35,077,44$GPZDA,164010.00,18,11,2016,,*68
    $GPGSV,4,4$GPRMC,164012.00,A,5044.9878947,N,00148.1095575,W,0.028,56.8,18111
    $G$GPRMC,164015.00,A,5044.9879113,N,00148.1095072,W,0.005,329.4,181116,0.0,E,D*2E

This isn't happening all the time but intermittently, which to me would imply a race condition: 

    $GPZDA,164009.00,18,11,2016,,*60
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49*7C
    $GPGSV,4,2,13,09,35,077,44$GPZDA,164010.00,18,11,2016,,*68
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49*7C
    $GPGSV,4,2,13,09,35,077,43,13,20,257,,06,19,195,,16,09,023,*79
    $GPG,N,00148.1095758,W,0.033,10.6,181116,0.0,E,D*1C
    $GPGSA,M,3,07,05,30,09,02,,,,,,,,3.8,1.7,3.5*3F
    $GPGGA,164010.00,5044.9878994,N,00148.1095758,W,9,05,1.7,3.806$GPZDA,164011.
    00,18,11,2016,,*69
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49*7C
    $GPGSV,4,2,13,09,35,077,44$GPZDA,164012.00,18,11,2016,,*6A
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,4,,23,02,079,
    ,49,32,171,40*73
    $GPGSV,4,4$GPRMC,164012.00,A,5044.9878947,N,00148.1095575,W,0.028,56.8,18111
    6,0.0,E,D*1B
    $GPZDA,164013.00,18,11,2016,,*6B
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49,09,023,*7E
    $GPGSV,4,3,13,29,06,306,,20,03,307,,23,02,079,,49,32,171,41*72
    $GPGSV,4,4,13,33,31,197,*46
    $GPRMC,164013.0064013.00,1.05,1.55,0.83,-14.9663,1.51,0.90,3.59*79
    $GPZDA,164014.00,18,11,2016,,*6C
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49*7C
    $GPGSV,4,2,13,09,35,077,43,13,20,257,,06,1M,0.037,N,0.069,K,D*2D
    $GPGST,164014.00,1.05,1.55,0.83,-14.9871,1.51,0.90,3.59*73
    $GPZDA,164015.00,18,11,2016,,*6D
    $GPGSV,4,1,13,07,62,087,51,30,59,164,51,05,54,296,47,02,37,232,49*7C
    $G$GPRMC,164015.00,A,5044.9879113,N,00148.1095072,W,0.005,329.4,181116,0.0,E,D*2E

Any idea what could cause this on iOS (using BLE) vs Android (using SPP)?

I've haven't changed the any of the [default UUID values in BLEDefines.h](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h#L31) for the Laird BL600, so could it be that I need to find and set different UUIDs for  the BL900?

don, 2016-11-18:
If you're communicating with the device, you shouldn't need to change anything in BLEDefines.

It's possible that I'm processing the data incorrectly or that the device is sending mangled data.

Are you streaming data from the device or downloading stored data?

dpa99c, 2016-11-18:
Thanks for the quick response.

Pretty sure the device isn't sending mangled data to iOS over BLE because I connected it with [Laird Serial](https://itunes.apple.com/gb/app/laird-serial/id663338164?mt=8) and the stream doesn't display the same mangling.

I'm streaming data from the device in real-time as it resolves positions via RTK/GNSS.

Is there anything I can do to help diagnose this? e.g. dump the raw data with subscribeRawData() ?

don, 2016-11-18:
Can you switch your code and try subscribeRawData on iOS?

If the data is not mangled, that points to a problem with the buffering in subscribe. 

How fast do the records arrive from the device?

dpa99c, 2016-11-18:
I'm out of the office now so don't have the device,  but I'll switch to raw data and post the results when I do. 

The data is arriving pretty damn fast. I'll dump out a timestamp with the raw data for reference.

dpa99c, 2016-11-21:
[Here's a raw data dump](http://ge.tt/7uWLDIh2) using subscribeRawData on iOS when connected to the Satlab SLC. 
I've echoed out the current time on each call to the handler function followed by the Uint8Array of bytes.
There's a couple of minutes of output and the messages seem to be arriving pretty fast, so there's about 4.5k lines of output.

dpa99c, 2016-11-21:
I've updated my use of subscribeRawData to convert the raw binary data back to strings:

    bluetoothSerial.subscribeRawData(function (data) {
        onSuccess(String.fromCharCode.apply(null, new Uint8Array(data)));
    }, onError);

But I'm still seeing mangling of the data, so subscribeRawData() is delivering the same mangling as subscribe(). Any suggestions?

dpa99c, 2016-11-30:
@don Any thoughts on this? 
Given that `subscribeRawData()` shows the same scrambling of serial data, do you have the feeling this could be an issue with the device itself, or with a race condition within the plugin?

don, 2017-01-05:
@dpa99c Sorry about the really long delay... 

Yes, since subscribeRawData shows the same scrambling of the data, it could be what the device is sending.

It's possible there's something getting munged in the plugin. It's managing 2 copies of the incoming data. The string and binary version.

You could try https://github.com/don/cordova-plugin-ble-central for a cleaner implementation.

You start a notification to register a callback that receives data 
https://github.com/don/cordova-plugin-ble-central/blob/master/examples/redbearlab/www/js/index.js#L79

Then your function gets called with an ArrayBuffer containing the data
https://github.com/don/cordova-plugin-ble-central/blob/master/examples/redbearlab/www/js/index.js#L87-L91

dpa99c, 2017-01-05:
@don Thanks for your response - I currently don't have access to the hardware to test this, but if and when I do, I will let you know the results.

don, 2017-01-06:
OK. Re-open this issue or create a new one if you have more info.

closed 2017-01-06 by don

dpa99c, 2017-01-06:
OK, fine. Will do.